RobotLocomotion/drake #12842: make `PiecewisePolynomial::breaks` public?

RussTedrake, 2020-03-10:
Is there a reason that it is not?  I was looking to plot the results of a trajectory optimization and was surprised that this information was not available to me.  My example is much less elegant without it.

Aside  -- we've never liked the name "breaks".  Might be a chance to change it before pushing to the public API.

RussTedrake, 2020-03-10:
Specifically, my code for getting the data out for plotting currently looks like this
```
x_sol = prog.ReconstructStateTrajectory(result)  # returns a PiecewiseTrajectory

times = np.linspace(x_sol.start_time(), x_sol.end_time(), 100)
x_values = np.empty((2,100))
for i in range(times.size):
  x_values[:,[i]] = x_sol.value(times[i])
```
yuck!  Would generally like to make that better.

ggould-tri, 2020-03-10:
The method `get_segment_times` on the superclass returns exactly `breaks_`, I think.  That might address both of your concerns.

RussTedrake, 2020-03-12:
Indeed.  My apologies.

closed 2020-03-12 by RussTedrake