RobotLocomotion/drake #12843: [geometry] Clean up internal documentation

SeanCurtis-TRI, 2020-03-10:
As per [GSG](https://drake.mit.edu/styleguide/cppguide.html#Doxygen), anything in an internal space should use either `//` or `/* */` so as to not suggest that it would appear in rendered doxygen.

Update all of the internal code to reflect that.

SeanCurtis-TRI, 2020-07-14:
@xuchenhan-tri This is the PR.

jwnimmer-tri, 2020-07-15:
For the record (so that I can LGTM the PR) -- in #13680 we discussed whether some kinds of comment markup within `geometry::internal` should be changed now that it's not processed by Doxygen.  We should conclude that discussion before closing this issue.

SeanCurtis-TRI, 2020-07-15:
I propose:

 - Any `@defgroup` or `@ingroup` should be right out. I think the arguments that those are defects in non-doxygenated documentation are strong and compelling.
   - The argument being, that `@ingroup` is an outright *lie* and `@defgroup` implies something that public documentation could link to. And that is neither true, nor do we want to encourage people attempting to link to the unlinkable.
 - I'm inclined to keep doxygen artifacts that are common in the code, even if they don't directly contribute to rendered documentation. E.g.,
   - `@file` can remain. If the documentation associated with the tag has value, then the tag makes its scope very explicit.
   - `@name` (and its attendant `//@{`, `//@}`) is worth keeping in terms of defining shared documentation and outlining the scope of the shared documentation.
  - Some artifacts don't directly contribute to communicating documentation concepts, but are innocuous in their impact and continue common lexical patterns as one reads the code. I wouldn't argue they must stay, only that they could stay. But I'm also open to arguments that they should go. E.g.,
     - % as prefix to class names.

The examples above are a very coarse sampling of the type tags in use. It might be worth creating a more exhaustive list because other tags may require more conversation. (E.g., I can see arguments forbidding our `@pydrake_mkdoc_identifier` as being strictly forbidden.)

Finally, anything that seems like it should be forbidden should make it to the styleguide.

SeanCurtis-TRI, 2020-09-25:
Resolved by #13680.

closed 2020-09-25 by SeanCurtis-TRI