RobotLocomotion/drake #12851: Using jupypter notebook for small quick interactive tools

huihuaTRI, 2020-03-11:
**Background:** We are developing a door hinge force element in Drake to provide the capability to model a door hinge joint, which generally exhibits complicate torque features. The solution is to consider five different torques and use different curves to approximate their characteristics. See [PR12812](https://reviewable.io/reviews/RobotLocomotion/drake/12812#-). 

**Problem statement:** Since there are many parameters to tune to model a particular hinge joint, the developers want to provide an interactive tool for the users to design the features of the hinge joint that they are interested in. Jupyter notebook would be a very good candidate for this requirement.  The problem is where should this jupyter notebook live. It would be preferred for it to live with the code. Would Drake support that? Another question is how could we point it to the reader/user that there is such a tool?

jwnimmer-tri, 2020-03-11:
> Would Drake support that?

What is the error message when you try it?

huihuaTRI, 2020-03-12:
It builds fine and I could run it with `bazel`.

jwnimmer-tri, 2020-03-12:
I guess in that case I'm not yet clear on what the bug report is about?  Are you asking in terms of what a policy should be?

huihuaTRI, 2020-03-12:
This is my understanding of the problem. Maybe @sherm1 has a different view.

Since it runs fine, it solves my concern. The discussion would be where to put this notebook. So, yes, it's about the policy.

jwnimmer-tri, 2020-03-12:
For my money, that is best discussed in the PR in that case.  It sounds like everyone was OK with it being adjacent to the new joint, but hearsay led someone to believe that the build system would fail in that case.

sherm1, 2020-03-12:
We need to hear from @hongkai-dai who was the source of the advice. I believe it had to do with installation rather than build.

ggould-tri, 2020-03-12:
Temporarily assigning @sherm1 (per platform review checklist) who can close or reassign as this clarifies.

hongkai-dai, 2020-03-13:
I think the current solution in the PR is fine? Previously we talked about whether you want to host the jupyter notebook online through binder. If we have to put it online then we need to teach the build system how to do it. I am happy to host it locally.

jwnimmer-tri, 2020-03-13:
> I think the current solution in the PR is fine?

In my opinion, I have a strong objection to the current PR which places one single class's specific documentation inside the tutorials folder.  If every class did that, we would drown with hundreds of tutorial files.  Tutorials should be reserved for information that is useful to wide swaths of our users, and which composes multiple classes and function together into unifying whole-program examples.  A little demo tool for one specific class is way too specific.

hongkai-dai, 2020-03-13:
> In my opinion, I have a strong objection to the current PR which places one single class's specific documentation inside the tutorials folder. If every class did that, we would drown with hundreds [...]

Agreed. This jupyter notebook is not for general tutorial, but more as an supplementary material for the door hinge class.

On a separate note, I am thinking should we bring some directory hierarchy to `tutorials` folder? For example, a `tutorials/solvers` folder for `MathematicalProgram`, a `tutorials/multibody` folder for MBP, etc.

jwnimmer-tri, 2020-03-13:
I don't mind extra structure there, but only 9 so far isn't _too_ big.  If I needed folders, I'd match Drake's homepage https://drake.mit.edu/ so "systems, solvers, multibody".

But even just some file renames without more folders might be good enough?  Having it all in one place makes it trivial to skim the list of tutorials without knowing ahead of time which section you should be browsing.

Or maybe the README.md file in that directory should contain a Table Of Contents that organizes the files into structure beyond just the alphabetical list.

huihuaTRI, 2020-03-13:
Since we have a consensus that the notebook should stay with the source code, I will change the PR and add some comments to instruct the user how to use it.

closed 2020-03-16 by huihuaTRI