RobotLocomotion/drake #12857: consolidate drake::trajectories and drake::systems::DenseOutput

RussTedrake, 2020-03-12:
Per a [conversation on slack](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1579047142005600), we would like `DenseOutput` to be a `Trajectory`, and to potentially migrate some of the nice features from `DenseOutput` up the class hierachy.  In that discussion, @hidmic wrote:
>  It seems there wasn't a concrete discussion involving drake::trajectories::Trajectory , but I clearly knew about them because I make heavy use of drake::trajectories::PiecewisePolynomial in https://github.com/RobotLocomotion/drake/pull/8950, and I even extended its API in https://github.com/RobotLocomotion/drake/pull/8949. Looking at both drake::systems::DenseOutput  and drake::trajectories::Trajectory , I'm under the impression the former is its own thing because the latter wasn't general enough. For a Trajectory , time is always a double (what if it is AutodiffXd ?) and values are always MatrixX<T>  (what if I want to optimize performance for a pure scalar dense output?). It was and is a purely theoretical caveat, as current implementations use ExtractDoubleOrThrow  and PiecewisePolynomial under the hood. That's as much as I recall.

sammy-tri, 2020-03-19:
>  we would like `DenseOutput` to be a Trajectory

(from old Slack convo):

>   It seems relatively straightforward to implement the Trajectory interface at the DenseOutput level.  Implementing Clone through the stack would be a pain, but perhaps it's ok to throw with a TODO for now (??).  The Trajectory base class also asks for derivative support, but I think that's a defect -- not all trajectories will have well-defined derivatives.

Basically agree with the above.  I don't think `Clone` would necessarily be so bad, every path I can currently find through the derived classes of `DenseOutput` looks like it finally reaches a `HermitianDenseOutput` which looks like it could get `Clone` in a straightforward way.  Since `HermitianDenseOutput` uses `PiecewisePolynomial` under the hood, the implementation of `MakeDerivative` is probably also straightforward though your comment about that being a questionable bit of the API is still relevant here and I'm not sure if it's worth bothering to expose for trajectories which are `DenseOutput`.  

>  to potentially migrate some of the nice features from DenseOutput up the class hierachy.

Which features were you thinking of?  At least at the `DenseOutput` API level the main one I see if evaluating only one scalar element, which would probably be doable, at least for vector valued trajectories.

I don't have my head wrapped around what well formed `AutoDiffXd` support looks like yet.  Given that a `HermitianDenseOutput<AutoDiffXd>` appears to discard the derivatives when building it's trajectory I'm not sure I need to think about it extensively right now.

RussTedrake, 2020-03-22:
Basically, I think we should have one set of trajectory classes... if there are new trajectory types in the denseoutput hierarchy, I would like to be able to use them in all of the use cases of trajectory.

But looking again now in order to answer your question, I think now that the entire premise of the `HermiteDenseOutput` class might be redundant?  Why is it not just a `PiecewisePolynomial`, and constructed using [`PChip`](https://github.com/RobotLocomotion/drake/blob/master/common/trajectories/piecewise_polynomial.h#L285)?

cc @hidmic

RussTedrake, 2020-03-23:
Just looked a little more closely.  I think the right solution is to have `InitialValueProblem<T>::DenseSolve` return a `Trajectory`.  I think that instances of `HermitianDenseOutput` should be replaced with `PiecewisePolynomial`, and instances of `DenseOutput` should be replaced by `Trajectory`.  The entire `DenseOutput` hierarchy should be removed.

There are definitely some features in `HermitianDenseOutput` that we'll need to move to `PiecewisePolynomial` -- like the incremental construction done in `IntegratorBase::DoDenseStep`.  But that's the whole point -- we have two nearly but not quite identical class hierarchies that need to be merged.

(fwiw -- The specific pain point that has been blocking me is that I need to use the output of an `InitialValueProblem` as the coefficients of of a `TimeVaryingLinearSystem` in order to implement the `TimeVaryingLinearQuadraticRegulator` properly; I could of course hack a conversion, but this is just the first of many places where I want to use the output of a simulation as a `Trajectory`.).

RussTedrake, 2020-03-24:
I do think the name `Hermitian__` is probably better than `PChip` for the `PiecewisePolynomial` method.  I hope we can take the best from both hierarchies.

RussTedrake, 2020-03-25:
I've been thinking about this enough now that I will take a first cleanup pass.

RussTedrake, 2020-04-07:
#13008 represents the final PR in the first phase of this cleanup -- the integrators no longer depend on `DenseOutput` (they use `PiecewisePolynomial` instead).   The only remaining consumer that I am aware of is the `InitialValueProblem` pipeline.  I think the `DenseOutput` hierarchy should now be deprecated.  The only question is whether we port the `InitialValueProblem` classes to use `Trajectory`, or update them, as well.

I think the functionality of `InitialValueProblem` -- making it easier to simulate a simple ODE described by. a function handle -- is great.  But I dislike the name (isn't every simulator solving an initial value problem?) and the fact that it re-implements many known concepts (`Context`, etc) and doesn't share it's nicer features with the rest of the drake system's framework.  My proposal would be:
  -  `internal::OdeSystem` in `initial_value_problem.cc` move to `systems\primitives` as e.g. `OdeFunctionHandleSystem`.  That is a generally useful tool.
  - Sugar on top of `Simulator` to replicate the `InitialValueProblem` workflow.  Port some of the tests.
  - Possibly port antiderivative to this alternative.  Would have to look more at the implications.
  - deprecate `InitialValueProblem`.

So we we let `DenseOutput` hang around for now until we can deprecate them both?
@sherm1, @sammy-tri , @jwnimmer-tri  -- what do you think?

sherm1, 2020-04-07:
Thoughts on this, @edrumwri ?

jwnimmer-tri, 2020-04-07:
My only comment is that once we have a plan in place, I would like to announce it in the release notes -- even if the deprecation markers have not yet been inserted.  I'd link back to this issue so users can become aware ahead of seeing the warnings.