RobotLocomotion/drake #12859: URDF parser doesn't load joint effort limit

sammy-tri, 2020-03-12:
The Multibody SDF parser loads joint effort limits, but the URDF parser lacks this feature.  It may be slightly more complicated in the URDF case because the effort is specified in the `joint` tag but the joint actuator is created when parsing the `transmission` tag.

sammy-tri, 2020-03-12:
Self-assigning for now.  If someone wants to work on this more urgently feel free to reassign.

sherm1, 2020-03-12:
Do we have a way to enforce the effort limit in MBP? It comes in with the joint but I guess it has to be applied to the MBP actuator input port?

sammy-tri, 2020-03-12:
We don't have a way to enforce the limit in MBP.   I opened this issue in response to a discussion with @yangwill on Slack, he confirmed for me that his interest right now is only in being able to query the value (which I believe would be basically equivalent to the parser support for joint velocity, which I don't think is enforced anywhere either).

sherm1, 2020-03-12:
I see. Getting it read in does seem like a good first step!

PR #12863 (2020-03-17): Parse joint effort limits from URDF
Fixes #12859

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12863)
commit 959138e1 (2020-03-17): Parse joint effort limits from URDF (#12863)

Fixes #12859
closed 2020-03-17 by SeanCurtis-TRI