RobotLocomotion/drake #12867: MathematicalProgram: Provide a common api for setting verbose solver output

RussTedrake, 2020-03-13:
If I want to debug my solver, I need to dig in and
1) figure out what solver is being called,
2) instantiate it myself,
3) figure out the particulars of this solvers options which sets the output.

Can we just have `prog.set_solver_output_level()' or something that provides a common interface for all of the solvers?  Possibly taking a filename as an argument, if that's desirable.  I know they are all a little different, but if we can unify them, it would be a real service!

jwnimmer-tri, 2020-03-13:
Though I can't find a citation now, in the past I recall talking about enhancing SolverOptions to allow for setting common options that would apply across all (or at least most) solvers, using a Drake-defined vocabulary, instead of the solvers' individual vocabularies.  That seems like the right path for this one.  The verbosity is a property of the Solve, not of the Program.

hongkai-dai, 2020-03-13:
> The verbosity is a property of the Solve, not of the Program.

Sorry I don't quite get it, `Solve` is a function, not a class, it cannot store a property? I think the print level should be in `SolverOptions`?

I think we can add a function
```C++
void SolverOptions::SetPrintLevel(int print_level, const std::optional<std::string>& file_name=nullopt)
```
Inside this function, it sets the print level for all supported solvers. Since `MathematicalProgram` also stores a SolverOptions object, we can also add the function 
```C++
void MathematicalProgram::SetPrintLevel(int print_level, const std::optional<std::string>& file_name=nullopt) {
  solver_options_.SetPrintLevel(print_level, file_name);
}
```
@jwnimmer-tri what do you think?

jwnimmer-tri, 2020-03-13:
Your general idea is consistent with mine, I just didn't express carefully.  I meant that the desire to print verbosely is related to the execution of a Solve, and not intrinsic to a program's algebraic formulation, so yes we should use SolverOptions to convey the desire to debug.

Whatever API we add to SolverOptions, I do not want to duplicate into MathematicalProgram.  (In fact, I want to remove `mp.SetSolverOption(...)` and just have `mp.get_solver_options().SetOption()` instead, so that we don't have to copy-paste the entire SolverOptions API surface into MathematicalProgram and keep both in sync.)  So I would like to choose an API that makes sense for SolverOptions on its own.

What is the preferred output format for the verbose solver output?  Should it just go to the console, unadorned?  Or should it use `drake::log()` instead, and so have severity information (info vs debug vs trace), and the ability for downstream code to reroute it through logger configuration?  Should it offer a filename option?  I would like to start from what the user wants, and then work our way back to what the solvers can do.

I'm not sure that I want to offer a `print_level` unless we reuse the spdlog levels -- defining a novel set of integer level semantics seems like extra work with little benefit.  If we just always dump everything that we don't need a level at all.  If we want offer levels, then let's explore if we can reuse spdlog's definitions for them.

hongkai-dai, 2020-03-14:
> Your general idea is consistent with mine, I just didn't express carefully. I meant that the desire to print verbosely is related to the execution of a `Solve`, and not intrinsic to a program's algebraic formulation, so yes we should use `SolverOptions` to convey the desire to debug.
> 

Agreed
> Whatever API we add to `SolverOptions`, I do not want to duplicate into `MathematicalProgram`. (In fact, I want to remove `mp.SetSolverOption(...)` and just have [...]

Makes sense. I also like to make `mathematical_program.h` smaller.
> What is the preferred output format for the verbose solver output? Should it just go to the console, unadorned? Or should it use `drake::log()` instead, and so have severity information (info vs [...]
Many solvers (Mosek, Snopt, etc) provide the API to print either to the console or print to a file. I think as a first pass I would like to emulate these solvers, to either print to console (if `file_name=nullopt`) or print to a file.

> I'm not sure that I want to offer a `print_level` unless we reuse the spdlog levels -- defining a novel set of integer level semantics seems like extra work with little benefit. If we just always [...]

Makes sense, how about this
```C++
void SolverOptions::SetPrint(bool print_flag, const std::optional<std::string>& file_name=null_opt);
```
We can only turn on or off the print flag.

hongkai-dai, 2020-03-16:
@jwnimmer-tri and I had a discussion on this, and here is our proposal

- Add a new file `drake_solver_options.h`, in this file we define three enum classes `DrakeSolverOptionsStr`, `DrakeSolverOptionsInt`, `DrakeSolverOptionsDouble`. And these enum classes have fields like
   ```C++
   enum class DrakeSolverOptionsStr {
     kPrintFile;
   };
   enum class DrakeSolverOptionsInt {
     kPrintToConsole;
   }
   ```
   These enums will be used as keys in `SolverOptions` class, to store the mapping from drake options to the value of the options. `solver_options.h` will include this `drake_solver_options.h` file.
- Inside `SolverOptions`, we will create three maps to store options that are common to all solvers
   - std::unordered_map<DrakeSolverOptionsStr, std::string> drake_options_str_;
   - std::unordered_map<DrakeSolverOptionsDouble, double> drake_options_double_;
   - std::unordered_map<DrakeSolverOptionsInt, int> drake_options_int_;
- We will also have the getter and setter for these three unordered_map objects.
- For convenience, we can provide two functions
   ```C++
   void SolverOptions::SetPrintFileName(const std::string& file_name);
   void SolverOptions::SetPrintToConsole(bool flag);
   ```
   These two functions just set the corresponding fields in `drake_options_str_` and `drake_options_int_`.
- In each of the solver (`mosek_solver.cc, snopt_solver.cc`, etc), we query the drake options, and call the corresponding solver-specific functions (to turn on/off print, etc).

jwnimmer-tri, 2020-03-16:
Yes, mostly.

The sugar functions allow for the first line here, as a replacement for the second:
```c++
const std::string foo = ...;
options.SetPrintFileName(foo);  // sugar
options.SetDrakeOption(DrakeSolverOptionsStr::kPrintFile, foo);  // no sugar
```

Given that the need for either formulation of this statement is rather rare, I'd prefer to err on side of less API duplication and not add the sugar function.

If anything, I would work to reduce the second line's boilerplate, maybe:
```c++
options.SetOption(DrakeSolverOptionStr::kPrintFile, foo);
```

Or if we wanted to move the enum-matches-type check to runtime instead of compile time, we could have only one enum:
```c++
options.SetOption(DrakeSolverOption::kPrintFile, foo);
```

That's the one I like the best.

hongkai-dai, 2020-03-16:
> Yes, mostly.
>
> The sugar functions allow for the first line here, as a replacement for the second:
>
> ```c++
> const std::string foo = ...;
> options.SetPrintFileName(foo);  // sugar [...]

Agreed, so we will have an enum class `DrakeSolverOption`, but three different unordered map as
std::unordered_map<DrakeSolverOption, std::string>, std::unordered_map<DrakeSolverOption, int>, std::unordered_map<DrakeSolverOption, double> inside `SolverOptions` class. I agree for the users it is easier to use one function `SolverOptions.SetOption()`.

jwnimmer-tri, 2020-03-16:
Yes.  Internally we can have a table with allowed type(s) for each option, and still fail-fast if the user specifies something incorrectly.

RussTedrake, 2020-03-17:
catching up on this.  I like the (final?) proposal.

jwnimmer-tri, 2020-05-21:
We have initial work ready with [CommonSolverOption](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1solvers.html#ae07229124cd16e9d8b62fe359c7d0b40) defined.  Is the next step to make a checklist of all solvers and implement this feature on each one by one?  Once we have one example of an implementation (e.g., for snopt), it should be easy for the community to contribute more implementations for other solvers?

hongkai-dai, 2020-05-21:
Thanks for the reminder, I forgot I had the PR open.

I can resume working on this later. My understanding is that this feature is less urgent than getting the dual solution. I would like to finish the dual solution PRs first (to get dual solutions for Gurobi, Ipopt, Snopt and SCS). Please let me know if I need to change my workflow.

jwnimmer-tri, 2020-05-21:
Fine by me.  I just stumbled into here through a keyword hit on a different issue.

RussTedrake, 2021-07-25:
Here is my attempt at the list that @jwnimmer-tri proposed we make to move this forward.  TTBOMK, the following solvers represent the set that do not yet respect `CommonSolverOption`.  I've ordered them by my personal bias for priority.
- [ ] Snopt
- [ ] Gurobi
- [ ] dReal (fyi @soonho-tri)
- [ ] CSDP
- [ ] OSQP
- [ ] IpOpt
- [ ] nlopt

hongkai-dai, 2021-07-27:
I think CSDP already respects CommonSolverOption::kPrintToConsole as shown in https://github.com/RobotLocomotion/drake/blob/e895d85fbcaf976cb8e41f6e9395ceb5c0ee576b/solvers/csdp_solver.cc#L418-L420 and tested here https://github.com/RobotLocomotion/drake/blob/e895d85fbcaf976cb8e41f6e9395ceb5c0ee576b/solvers/test/csdp_solver_test.cc#L357-L363

SeanCurtis-TRI, 2021-07-28:
I'm tackling OSQP and I'm pretty sure I've fallen at the first post.

1. It appears OSQP doesn't provide settings for writing to a file. Looking [in the header](https://github.com/osqp/osqp/blob/master/include/types.h#L161-L167), we only have the `int` indicating verbosity (which I presume should be associated with `kPrintToConsole`). So, what protocol do we have when `kPrintFileName` is set but can't be used?
2. This issue seems to be missing the [protocol for resolving options](https://github.com/RobotLocomotion/drake/pull/15257#pullrequestreview-695385202) (as defined in #15257). The protocol is reproduced here for convenience:
    - The priority given to solver options is as follows (from lowest / least, to highest / most):
        - common option set on the MathematicalProgram itself
        - common option passed as an argument to Solve
        - solver-specific option set on the MathematicalProgram itself
        - solver-specific option passed as an argument to Solve

This protocol seems relevant because conflict between `MathematicalProgram` options and `Solve` options is resolved in the `SolverBase` (via merging the options). But each solver is independently responsible for resolving between common and solver-specific configurations.

jwnimmer-tri, 2021-07-28:
> What protocol do we have when kPrintFileName is set but can't be used?

_"If the solver doesn't support the [common] option, the option is ignored."_

-- https://github.com/RobotLocomotion/drake/blob/v0.32.0/solvers/solver_options.h#L80-L85

SeanCurtis-TRI, 2021-07-28:
Lastly, testing the functionality for writing to a file is straightforward. Do we have a desired practice for testing writing to console?

hongkai-dai, 2021-07-28:
> Lastly, testing the functionality for writing to a file is straightforward. Do we have a desired practice for testing writing to console?

No we don't test what has been written to the console, but we test printing to the console doesn't cause a runtime error.

jwnimmer-tri, 2022-04-26:
Here's the full list of implementation tasks.  (To be updated as we go.)

- [ ] ClpSolver to console
- [ ] ClpSolver to file
- [ ] CsdpSolver to console
- [ ] CsdpSolver to file
- [ ] DrealSolver to console
- [ ] DrealSolver to file
- [ ] EqualityConstrainedQPSolver to console
- [ ] EqualityConstrainedQPSolver to file
- [ ] GurobiSolver to console
- [ ] GurobiSolver to file
- [ ] IbexSolver to console
- [ ] IbexSolver to file
- [ ] IpoptSolver to console
- [ ] IpoptSolver to file
- [ ] LinearSystemSolver to console
- [ ] LinearSystemSolver to file
- [ ] MobyLCPSolver to console
- [ ] MobyLCPSolver to file
- [ ] MosekSolver to console
- [ ] MosekSolver to file
- [ ] NloptSolver to console
- [ ] NloptSolver to file
- [ ] OsqpSolver to console
- [ ] OsqpSolver to file
- [ ] ScsSolver to console
- [ ] ScsSolver to file
- [ ] SnoptSolver to console
- [ ] SnoptSolver to file
- [ ] UnrevisedLemkeSolver to console
- [ ] UnrevisedLemkeSolver to file

PR #12884 (2020-03-23): add DrakeSolverOption
This is the first step to resolve #12867 . This PR adds a new enum class `DrakeSolverOption`, and the corresponding functions in `SolverOptions` class.

I will add the python binding, and the function in each solver to handle `kPrintFileName` and `kPrintToConsole` in the future PRs.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12884)