RobotLocomotion/drake #12868: Improved CI support for programs using X

jwnimmer-tri, 2020-03-13:
In PR #12788, we have some rendering code that is unable to run using xvfb in CI.  I don't have the exact errors handy but hopefully @tehbelinda and @SeanCurtis-TRI can help point us to examples.

They are going to try to rework the code to pass with xvfb, but failing that would like to request that Ubuntu CI offer a more comprehensive graphics stack.

In TRI's repository, I have code to launch xorg service.  I'll make that available shortly.

See also related from the past:  #2917, #5227, and some others in the same family.

jamiesnape, 2020-03-16:
One question is are we are going to need instances with discrete GPUs as well? If so, that would mean cutting some jobs to stay within budget.

jwnimmer-tri, 2020-03-16:
My mental model of the project says that dGPU will not be required.  All of the current unit tests pass in Anzu when run on laptops with integrated graphics.  But @SeanCurtis-TRI would be authoritative.

jwnimmer-tri, 2020-03-16:
FYI I posted my Anzu current CI snippet at https://gist.github.com/jwnimmer-tri/8e28f3ea117e8ee543d52e94a9948237.  The "Device" section is clearly for a dGPU host (EC2 G3 series), but hopefully could be tweaked for C4 or whatever Drake CI is using.

SeanCurtis-TRI, 2020-03-16:
At this point, we're using vanilla OpenGl -- we should be fine for a long time.

tehbelinda, 2020-03-16:
Here's an example of a CI error:
```
Using drake_cc_googletest_main.cc

[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from OpenGlContext
[ RUN      ] OpenGlContext.GetContext
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  34 ()
  Serial number of failed request:  28
  Current serial number in output stream:  26
```

jwnimmer-tri, 2020-03-16:
> They are going to try to rework the code to pass with xvfb, ...

Is there any incremental progress to report yet?  Or updated forecast whether or not it is likely to succeed?

SeanCurtis-TRI, 2020-03-16:
None from me. @tehbelinda?

tehbelinda, 2020-03-16:
Looking at VTK, I think it works in CI because it uses GLEW instead? I'm looking into whether we can use that instead

jamiesnape, 2020-03-17:
GLEW would make the code more robust, certainly.

From a cursory reading of the code, looks like you need OpenGL 3.0-3.2 with the particular code you are using, but I do not necessarily see any functionality that needs OpenGL 3.  As I said, though, it was a cursory glance, so I may be wrong.

Regardless, the higher the version of OpenGL you need, the greater chance you have of making users unhappy. So something to balance, irrespective of Xvfb.

EricCousineau-TRI, 2020-03-19:
@tehbelinda Curious, is this on `master`? We have code that runs find with Xvfb in Anzu, and does things like this. I will send you and Sean a Slack message to point out where.

jwnimmer-tri, 2020-03-19:
The code is all in PRs: https://github.com/RobotLocomotion/drake/pulls/tehbelinda

EricCousineau-TRI, 2020-03-19:
Aye, so encountering it here effectively:
https://github.com/RobotLocomotion/drake/blob/3f7c6520a985e28282cf5498a3b897cefdbea1b9/geometry/render/gl_renderer/opengl_context.cc#L40

tehbelinda, 2020-03-20:
I tried the GLEW/GLUT combo, but then ran into the same problem because it turns out GLUT still uses X under the hood to get a window context, oops.

So then I tried using GLEW/OSMesa instead for pure off-screen rendering, based off VTK's [vtkOSOpenGLRenderWindow](https://github.com/Kitware/VTK/blob/master/Rendering/OpenGL2/vtkOSOpenGLRenderWindow.cxx), but am unable to use the context it returns. glewInit fails with Missing GL version, and some other gl calls, like creating a shader, cause a seg fault. I'm thinking either one of two scenarios: 
1. there is more behind the scenes VTK magic and it's not actually using Mesa either, or 
2. there is something on my local system that is getting in the way. (I thought it might be nvidia so I tried setting up an alternate display to test it with mesa, but still no luck)
```
belindateh@T480-BTeh:~$ Xephyr -br -ac -noreset -screen 1920x1080 :1&
[1] 29002
belindateh@T480-BTeh:~$ glxinfo | grep OpenGL | grep string
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce MX150/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 435.21
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 435.21
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 435.21
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
belindateh@T480-BTeh:~$ DISPLAY=:1 glxinfo | grep OpenGL | grep string
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 9.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 19.2.8
OpenGL core profile shading language version string: 3.30
OpenGL version string: 3.1 Mesa 19.2.8
OpenGL shading language version string: 1.40
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 19.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
```

I'm getting less hopeful now...I might try EGL next, but am starting to think launching the service so it's more like the Anzu CI is going to be the easier way forward

jamiesnape, 2020-03-20:
> I tried the GLEW/GLUT combo, but then ran into the same problem because it turns out GLUT still uses X under the hood to get a window context, oops.

No reason GLUT could not work. You do have X already, you just have fewer OpenGL features and extensions.

tehbelinda, 2020-03-20:
Hmm but in CI I still saw the same X error? https://drake-cdash.csail.mit.edu/testDetails.php?test=37519443&build=1275560

tehbelinda, 2020-03-20:
Oh I see what you mean, if I reduce the OpenGL version to 3.3...ok let me try further with this

jamiesnape, 2020-03-23:
VTK uses OpenGL 2.x, for instance. If you stick to that, you will certainly be ok. The laptop I am currently using, quite new, but using Intel UHD graphics, only supports OpenGL 2.1. That is not an uncommon situation for the average laptop.

tehbelinda, 2020-03-23:
Ok [CI](https://drake-jenkins.csail.mit.edu/job/linux-bionic-unprovisioned-gcc-bazel-experimental-release/54/) seems happy with 3.3 - is it possible to get a new provisioned build with glut? @jamiesnape

jamiesnape, 2020-03-24:
> Ok [CI](https://drake-jenkins.csail.mit.edu/job/linux-bionic-unprovisioned-gcc-bazel-experimental-release/54/) seems happy with 3.3 - is it possible to get a new provisioned build with glut? @jamiesnape

I need at least one LGTM on a PR that adds GLUT to the provisioning script(s) (#12788 or separate PR are both fine) and an issue asking for the provisioning that is assigned to myself and `@BetsyMcPhail`.

tehbelinda, 2020-03-26:
Solved by reducing the openGL context version down from 4.5 to 3.3

closed 2020-03-26 by tehbelinda