RobotLocomotion/drake #12877: Add SolveWithId on top of Solve function

hongkai-dai, 2020-03-15:
Currently to solve a mathematical program with a specified solver, we have to instantiate that solver, and the call `solver.Solve(prog)`. This works find when the users have access to `prog`. On the other hand, when the user doesn't have access to `prog`, this workflow breaks. Consider the following example

```C++
void foo() {
  MathematicalProgram prog;
  prog.AddConstraint(...);
  Solve(prog);
}
```
Now the user cannot specify which solver to be used inside `foo` function. `Solve` function would choose the solver automatically.

There are some cases that the user wants to specify which solver to use (for example, in #12876 , the default solver has numerical issues). I think the easiest way for the user is to provide a list of `SolverId`s, and then we have a function to choose the first available solver with the matching ID. Namely we can have the following function
```C++
void SolveWithId(const MathematicalProgram& prog, const std::vector<SolverId>& solver_ids, const std::optional<VectorXd>& x_initial, const std::optional<SolverOptions>& solver_options);
```

cc @RussTedrake

jwnimmer-tri, 2020-03-16:
I would counter-propose two things:

(1) New function
```c++
std::vector<SolverId> ChooseBestSolvers(const MathematicalProgram& prog);
```
We could return _all_ compatible solvers in preferential order.  Then users who desire extra logic in refining what "best" means could post-process the list, in order to choose the one they like.  The selection heuristic could be arbitrarily complicated, because we've left it to the user.  (The user could also do something like fire off _all_ of the compatible solvers in different threads -- the "all" here vs "best one only" previously seems like nice new information.)

(2) New function with "excluded" option:
```c++
SolverId ChooseBestSolverExcluding(
  const MathematicalProgram& prog,
  const std::set<SolverId>& excluded);
```
That would cover the common case given in this issue, where on particular solver is known to misbehave on a certain program.

Note that in both cases, I would prefer to keep the surgery isolated to a `Choose...` function.  The `Solve()` function is just two lines of sugar, and I don't see a need to thread these customizations through to the sugar -- at least not until we find ourselves employing them frequently.

```c++
void foo() {
  const SolverId solver_id = ChooseBestSolverExcluding(prog, {MosekSolver::id()});
  MathematicalProgramResult result = MakeSolver(solver_id)->Solve(prog);
  if (result.is_success()) {
    // ...
  }
}
```

jwnimmer-tri, 2020-03-16:
(3) Another new proposal: the user dictates their preferred order, and we help filter:

```c++
std::vector<SolverId> FilterPossibleSolversFrom(const std::vector<SolverId>& preferential_order)
SolverId ChooseFirstSolverFrom(const std::vector<SolverId>& preferential_order) {}
```

Filtering based on available() and AreProgramAttributesSatisfied().

hongkai-dai, 2020-03-16:
@RussTedrake If we want to specify a solver, we could do
```C++
void RoA(SolverId solver_id) {
  auto solver = solvers::MakeSolver(solver_id);
  auto result = solver->Solve(prog);
}
```
Could you talk more about the use case when passing a list of solver IDs? Do you want to exclude some solvers? Do you want to try the first of these solvers, and then try the second solver if the first one fails?

RussTedrake, 2020-03-17:
Unfortunately, I think there are two use cases.  

1. Most **users** will feel like they know what solver that they have installed, and want to be using right now.  This is the case that we should make easy for people (including non-experts), and should throw if the solver is not capable of supporting that problem.
2. Developers who want to commit code to drake need to think more about the matrix of supported solvers.  To add an optimization routine to drake master, I agree that we would like an API that will automatically support new solvers if/when they are introduced, and black-list a solver that we know to fail.  But I can also imagine an actual priority list .. e.g. for this algorithm I would use Mosek if you have it, else Gurobi, else CSDP, else whatever solver MP would have recommended.  I could imagine being happy with whichever API for this you and @jwnimmer-tri  recommend.

jwnimmer-tri, 2020-03-17:
I think Russ's (1) is fully supported and already has appropriate sugar.  (In fact I don't understand the top post's complaint at all.)  For example, here's how a caller can specify a solver to a hidden program:

```c++
void CalcFoo(SolverId id) {
  MathematicalProgram prog = ...;  // complicated, internal program
  auto result = MakeSolver(id)->Solve(prog);
  if (result.is_success()) ...
}
```

Or if we want to just hard-code a solver:
```c++
void CalcBar() {
  MathematicalProgram prog = ...;  // complicated, internal program
  auto result = SnoptSolver().Solve(prog);
  if (result.is_success()) ...
}
```

---

I think the discussion in the issue mostly gets at how to suport Russ's (2) with some nice new routines.

RussTedrake, 2020-03-17:
I did not realize we already had `MakeSolver(id)`.  That does help (1).

hongkai-dai, 2020-03-17:
I will add `MakeSolver(id)` to the mathematical program tutorial.

hongkai-dai, 2020-03-17:
Just added `MakeSovler(id)` to the tutorial in #12885

jwnimmer-tri, 2020-03-25:
FYI As we fix this, there is a helper function in Anzu that might be worth bringing over at the same time:
```c++
drake::solvers::SolverId GetSolverIdFromName(const std::string& name);
```

But even better than that function, I think the good API would be a simple function that returned a list of all `SolverId` that are built into Drake.  Then the user could write any predicates they wanted over that list -- e.g., searching by name, or adding / removing choices, filtering by `available` and/or `AreProgramAttributesSatisfied`, etc.

The key pieces that the user does _not_ yet have are (a) the list of all solvers, and (b) the ranking of "best".  It's easy to fix (a) quickly.

jwnimmer-tri, 2021-11-11:
> Developers who want to commit code to drake need to think more about the matrix of supported solvers. To add an optimization routine to drake master, I agree that we would like an API that will [...]

Please hyperlink to the places in Drake (lines of code) that would benefit from some new sugar -- where Drake would like to customize it's internal call sites into solvers to tweak the solvers' priority order.  Once we have that, we can figure out which of the above ideas (or a new idea) would be the best resolution for those practical examples.

RussTedrake, 2021-11-12:
Here we have the users unable to configure the solver (but we would like them to); they must use the the one determined by ChooseBestSolver:
https://github.com/RobotLocomotion/drake/blob/ef943c03da077e2acbd556d0b52bcf20475cafd1/geometry/optimization/graph_of_convex_sets.cc#L470

This example is one where the default solver gave bad answers, and for this problem specifically we would have liked to use another one:
https://github.com/RobotLocomotion/drake/issues/12876

This example is one that has been a pain through the years, since the success of snopt/ipopt has been so sensitive on the initial guess that it sometimes breaks when we update the solver version:
https://github.com/RobotLocomotion/drake/blob/ef943c03da077e2acbd556d0b52bcf20475cafd1/examples/acrobot/test/run_swing_up_traj_optimization.cc

hongkai-dai, 2021-11-12:
We have [MakeFirstAvailableSolver](https://github.com/RobotLocomotion/drake/blob/fd85e98a0626463dfdc40c4daaa838d44d06838e/solvers/choose_best_solver.h#L32-L37) that returns the first available solver from a list of IDs, the user can then call that solver for the program. An example code is
```cc
auto solver = MakeFirstAvailableSolver({CsdpSolver::id(), MosekSolver::id(), ScsSolver::id()});
auto result = solver.Solve(prog);
```
Would this solve @RussTedrake 's problem? The downside of this approach is that when calling `MakeFirstAvailableSolver` it doesn't know if the solver is capable of solving the mathematical program.

jwnimmer-tri, 2021-11-12:
On the first example ("1"):

> Here we have the users unable to configure the solver (but we would like them to); they must use the the one determined by ChooseBestSolver:
> https://github.com/RobotLocomotion/drake/blob/ef943c03da077e2acbd556d0b52bcf20475cafd1/geometry/optimization/graph_of_convex_sets.cc#L470

Let me make sure I understand this.  When a user calls `GraphOfConvexSets::SolveShortestPath` we want to allow them to customize which solver will be selected.  They could do this either via a new standalone configuration-setting function on `GraphOfConvexSets`, or by passing a new (optional) argument to `SolveShortestPath`, but I think for the purposes of this issue those are equivalent -- the question is what _kind_ of information we want the user to provide us.

Would you like the user to pass us, either:

(a) An ordered list of `SolverId`.  We find the first one that is available and compatible with the program, and then construct it and use it.  This means that only Drake's hard-coded suite of available solvers could be used.

(b) An ordered list of `const SolverInterface*`.  We find the first one that is available and compatible with the program, and then use it.  This would additionally allow for users to provide non-Drake solvers.

(c) A `SolverOptions` structure that will be used for the solve, with the new feature that choice-of-built-in-solver preferences are a new kind of get/set configuration knob on the existing options class.  This would mean that all calls to Solve automatically gain the ability to configure the preference order, not just the GOCS code and alike that opts-in to some special syntax.

(d) Something else?

For all of those options, if the user gives us a list and none of the items on the list are available and compatible, should our response be to throw an exception, or to _also_ search the ChooseBestSolver database in its entirety to find one that might work?

jwnimmer-tri, 2021-11-12:
On the second example ("2"):

> This example is one that has been a pain through the years, ...
> https://github.com/RobotLocomotion/drake/blob/ef943c03da077e2acbd556d0b52bcf20475cafd1/examples/acrobot/test/run_swing_up_traj_optimization.cc#L73-L74

I am not sure what improvement we're looking for in this case.  The program hard-codes Snopt.  Is the request to change it to call something more generic along the lines of directly using `solvers::Solve` or `solvers::ChooseBestSolver`, but with an easy way to disallow IPOPT as a candidate solver because know that IPOPT is unsuitable in practice here, even though the program attributes appear to be compatible at first glance?

jwnimmer-tri, 2023-09-23:
About two years ago, I posed questions upthread about whether this sugar is really necessary, what specific kinds of information / sugar we want to provide, and how widely it should apply.  By the lack to reply, I can only assume the sugar is not important, and we should either close this issue (or reply to the questions, so that the ticket is actionable).

hongkai-dai, 2023-09-23:
I think we can close this issue. I think `MakeFirstAvailableSolver` already satisfies a lot of the user needs.