RobotLocomotion/drake #12887: Test whether or not linear_bushing_roll_pitch_yaw is quasi-symmetric

mitiguy, 2020-03-17:
Per comments in PR# 11946, there is a need for a "reversibility" test.

@amcastro-tri commented about whether the quasi-symmetric LinearBushingRollPitchYaw class is symmetric in a specific way.  

@sherm1 responded with all the following:
Before we say any more about this, I would like to test the reversibility of the bushing. Although the parameterization is asymmetric, I think relabeling A and C and reversing the parameterization should yield exactly the same forces and torques. Something like this I think (prime here means reversed):

L0'=L1, A'=C, L1'=L0, C'=A
X_A'C' = X_CA (physical kinematics is unchanged)
V_A'C' = V_CA
xyz' = −xyz
rpy' = −ypr (note reversed order)
k012' = k210, d012' = d210
kxyz' = kxyz, dxyz' = dxyz
My speculation is that, given the above substitutions, we would find that F_A'=F_C and F_C'=F_A=−F_A' as before.

If that's true, I think we can say the bushing is physically symmetric and reversible when symmetrically parameterized."

amcastro-tri, 2020-03-18:
Further thought on this made me realize that @sherm1's idea to reverse the bushing should work.
What I realized also, I believe, is that the mechanical system represented still is not symmetric. What happens when you "reverse" it is that you recover the original non-symmetric model.

In other words, someone has to say `bushing_element.Reverse()` right? Again, not a defect but  trying to understand what we have here.

sherm1, 2020-03-18:
I don't understand your comment above, Alejandro. Are you saying you think the current bushing doesn't produce identical forces and torques at A and C? That is what we're claiming it does but it is easy to test.

amcastro-tri, 2020-03-18:
In a nutshell, someone has to say `bushing_element.Reverse()` by hand. What does that tell you about the physics being modeled?

sherm1, 2020-03-18:
Nothing -- I still don't understand your point. I think we're all in agreement that "quasi symmetric" refers to whether the forces and torques are the same on both bodies. That is all we're asking for in a "symmetric" bushing -- we know the parameterization is not symmetric. I can't tell from your comment whether you are saying that we aren't even achieving that limited kind of symmetry?

closed 2024-03-20 by jwnimmer-tri