RobotLocomotion/drake #12890: Document Drake-standard meanings for "accuracy" and "tolerance"

sherm1, 2020-03-17:
(From discussion in PR #12875)

We've been inconsistent in use of terms like "precision", "accuracy", and "tolerance" in Drake. We should document how we define these terms and standardize on usage. From the PR discussion:

> **Sherm**: FWIW I generally use "tolerance" as an absolute quantity with units, as in "constraint tolerance" which means the amount by which some error can deviate from zero. I use "accuracy" to mean a relative error for a quantity that is normally expected to be far from zero. Then "accuracy" can be used to estimate a tolerance: basically tol=accuracy*|quantity_under_test|. If the quantity under test happens to be near zero, we use a unit scaling to prevent an absurdly tight tolerance:
tol=accuracy*max(unit_value, |quantity_under_test|). In practice the unit_value is almost always set to 1 or 0.1 due to our nice choice of units.
> **Russ**: Should we capture something like this and add it to the developer / platform review guidelines? It was triggered because the examples matrix_util.h do feel inconsistent -- one uses "precision" and the other uses "tolerance".
> **Sherm**: Standardizing the terms and documenting them somewhere seems like a great idea. Where should that go? This seems closer to a notational convention than a coding guideline. We could add it to Modules/Multibody Kinematics and Dynamics/Terminology and Notation. The conventions there are already used more broadly in Drake than just the multibody stuff.
> **Russ**: I could imagine it living in a "terminology and notation" section of doxygen instead of sphinx, I guess. But I would never think to look for it there, precisely because it is not multibody related?
> **Eric C**: I'd definitely be down for it living in a general notation section! It could live at the same level as others, e.g. next to "Algorithms", "Technical Notes", "Multibody Kinematics and Dynamics", etc. If we wanted, that section could then link to Multibody notation / terminology and other more specific areas.

I like Eric's idea of a general Drake notation & conventions section at the top "Module" level in Doxygen. I propose as a first crack to add that Module, write the accuracy/tolerance section, and then link to the Multibody notation section from there. Later we could consider moving up some of the more-general material in the Multibody notation to the higher-level Module.

Regarding precision/accuracy/tolerance: I would propose we _not_ use "precision" for numerical programming but reserve it as an ML term of art (precision/recall). We should use accuracy and tolerance as I defined them above. The intent would be that misuse of these terms would be a revieable defect. Let me know if you disagree, otherwise I'll post a PR to that effect.

EricCousineau-TRI, 2020-03-17:
> Regarding precision/accuracy/tolerance: I would propose we not use "precision" for numerical programming but reserve it as an ML term of art (precision/recall)

:100: with you on this one!

amcastro-tri, 2020-03-26:
A very similar view in review comment of #12895:

> Another way to look at this is: tolerance = absolute_tolerance + relative_tolerance * abs(quantity), but you also happened to chose the same value, kTolerance_, for both absolute_tolerance and relative_tolerance, even though absolute_tolerance has units and relative_tolerance doesn't.

PR #13267 (2020-05-11): Add module for "accuracy" and "tolerance" conventions
Per issue #12890, this Doxygen-only PR does the following:
- Adds a top-level `Module` to collect Drake conventions for terminology and notation.
- Moves the existing "Multibody Terminology and Conventions" module under the new one.
- Adds a new module defining Drake's conventions and terminology for "accuracy" and "tolerance". 

The intent with the new section is that using these terms differently (or using other terminology like "precision" instead) would be a defect in new Drake code, and that there is now an `@ref` link we can use in Dr [... cut]
closed 2020-05-11 by EricCousineau-TRI