RobotLocomotion/drake #12892: Support `IntegratorBase<Expression>`

RussTedrake, 2020-03-18:
I am starting to try to write algorithms/systems that take an `IntegratorBase` as an argument.  For example, in `DirectTranscription`, I would like users to be able to choose the integration for continuous systems from the suite of integrators that we have.  But `DirectTranscription` also tries to add symbolic constraints when possible -- for instance so that `MathematicalProgram` can realize when a constraint is linear.  Ideally, this would work with the suite of integrators, too.

Another use case is my intention to write a general `DiscreteTImeApproximationSystem` that takes a continuous-time system and provides the abstraction, again using the suite of possible integrators.

After a little poking around, I think it's probably possible and even reasonable with a few well-placed `std::enable_if`'s and perhaps changing some uses of `T` back to `double`.  The fundamental challenge is that `IntegratorBase` has a lot of branching logic for the error-controlled integrators -- I don't think it's reasonable to expect those to work.  But I do think that fixed-step integrators should support symbolic.

Opening this issue to track it.

jwnimmer-tri, 2020-03-18:
These days the code changes should be even more minimal than years past.  We have `Formula` now contextually convertible to `bool` and `if constexpr (scalar_predicate<T>::is_bool) { ...` available as of C++17.

jwnimmer-tri, 2020-03-24:
@RussTedrake was #12922 sufficient or is there more to do here?

RussTedrake, 2020-03-24:
#12922 was sufficient for now.  We can add support for more integrators on demand.  Thanks!

closed 2020-03-24 by RussTedrake