RobotLocomotion/drake #12912: tutorials: Need to add tutorial for MultibodyPlant

EricCousineau-TRI, 2020-03-20:
At present, we have tutorials for formulating IK / MP problems and doing rendering with MBP.
However, we don't have any succinct tutorials covering how MBP works.

Some things to do (which could be done in stages):
- Intro (kinematics, dynamics, topology, etc). Mostly, this should point to Doxygen / Underactuated textbook, and not repeat too much?
- Parsing models. Should also cover parsing multiple models, and model instances.
- Querying joints, bodies, etc. Should cover model instances (e.g. errors you get when you have multiple of the same name).
- Minimal coordinates (positions, velocities). Read/write. Talk about floating-body base coordinates?
- Maximal coordinates (pose). Read/write. CalcRelativeTransform, etc.
- Visualization (relates #12645, though perhaps we could do 2D projections?), first w/o sim.
- Dynamics.
- Basic geometry, collision. (We may want to save complex bits for a separate tutorial?)
- Simulation. Sensors, force
  - Also cover purely kinematic simulations? (e.g. driving kinematics by input ports to position / velocity).

After reviewing [Zhaoyuan's tutorials](https://drake.guzhaoyuan.com/introduction/drake-multibody), I see some minor inaccuracies. Once we have the tutorial in Drake, perhaps we should try to get him to link to ours?

FYI @sherm1 @huihuaTRI @RussTedrake

RussTedrake, 2020-03-21:
I've had this on my list for a long time, too.  Here was my chicken-scratch thinking about what to include:
- Creating your multibody system from urdf/sdf.  Simple geometry inspector interface with jupyter joint sliders and an embedded meshcat visualizer
- Core design: MBP + SceneGraph
- Simulation.  
  - Visualizers.  Playback.
  - Simulator as a ROS node
- Kinematics and dynamics.  Show symbolic.
- (Inverse) kinematics.  Mention global IK
- Call out to the existence many other advanced algorithms (for motion planning, etc)

hongkai-dai, 2020-03-22:
I am modifying IK API as mentioned in https://github.com/RobotLocomotion/drake/issues/12849#issuecomment-598848762, I will add the IK tutorial after the API is stabilized.

EricCousineau-TRI, 2020-04-22:
Relates #13084

jwnimmer-tri, 2022-04-15:
See also #16643, which intends to implement a portion of the topics listed above.

jwnimmer-tri, 2022-05-06:
Closing as a duplicate of #16643.  I've copied all of the ideas above into that issue.

closed 2022-05-06 by jwnimmer-tri