RobotLocomotion/drake #12923: `isnan(Expression::NaN())` should return `Formula::True()`

RussTedrake, 2020-03-23:
Sorry if I’ve made a mistake, but I’m pretty sure that currently
```
static_cast<bool>(isnan(Expression::NaN()))
```
Throws an exception, because it has NaN in the environment.
I’d like it to return True.
(Once done, I can add `Expression` support for `RungeKutta3Integrator`)

jwnimmer-tri, 2020-03-23:
Note:
```c++
ExpressionConstant::ExpressionConstant(const double v)
    : ExpressionCell{ExpressionKind::Constant, true, true}, v_{v} {
  DRAKE_ASSERT(!std::isnan(v));
}
```

It is not allowed to place NaN into a scalar when `T = Expression`.  Because symbolic algebra with NaN (and symbolic formulae over NaN) have complicated rules and defeat any straightforward attempts at using implication to simplify trees as they are constructed, we have ruled it out entirely.

My take is that the integrators' use of NaN to indicate an unset value is the real problem here.  It's super confusing in the first place (the docs fail to warn users about NaN as a possible return value from the getters, for example).  I think we should purge this idiom from the integrators.

We can leave open the question of whether CalcTimeDerivatives is allowed to return NaNs to indicate an impossible step, but things like get_initial_step_size_target have no business returning NaN.

jwnimmer-tri, 2020-03-23:
Oops, the first part of what I said was wrong:
```c++
shared_ptr<ExpressionCell> Expression::make_cell(const double d) {
  if (d == 0.0) {
    // The objects created by `Expression(0.0)` share the unique
    // `ExpressionConstant` object created in `Expression::Zero()`.
    //
    // See https://github.com/RobotLocomotion/drake/issues/12453 for details.
    return Expression::Zero().ptr_;
  }
  if (std::isnan(d)) {
    return make_shared<ExpressionNaN>();
  }
  return make_shared<ExpressionConstant>(d);
}
```

We promote NaN to its own cell type, so the constructor I posted was not relevant.

So yes, I think its valid to look into whether `isnan(Expression(NaN))` could behave better.

However, for the integrators, I still think we should remove the footgun sooner or later, even if we happen to resolve this issue's feature request first.

jwnimmer-tri, 2020-03-23:
It turns out we have unit tests spelling out exactly what happens here:
https://github.com/RobotLocomotion/drake/blob/d78538c3cd6a3a172f641b0ef4b03ce7ae34f2ed/common/test/symbolic_formula_test.cc#L275-L289

RussTedrake, 2020-03-23:
@soonho-tri is awesome.  :-)
But I am surprised at that throwing being considered the desired behavior. 

I agree that the pattern with NaNs in `IntegratorBase` is not ideal.   @jwnimmer-tri -- can you open an issue for it?  (I think you will state the problem with more conviction than me).

jwnimmer-tri, 2020-03-23:
I believe the NaN handling behaviors came from lots of f2f discussions between @soonho-tri and myself(?).

I think the reasoning was like so:
1. NaN values are extremely rare during typical computations.  Because they are difficult to handle symbolically, we'll round that up to "must never occur".  We allow the user to form `ExpressionNaN` cells in a symbolic tree, but evaluating a tree that has them _anywhere_ in its structure is an error (modulo rule 3 below).
1. It's still valid for code to check `isnan` in order to fail-fast, though -- so we should provide `isnan(Expression)` for the common case of non-NaN value returning `False`.  This way, code can fail-fast with double yet still compile with Expression (test case L279).
1. If there are expressions that embed separate cases (`if_then_else`), some of the sub-expressions may be invalid when they are in the not-taken case (for NaN reasons or any other reason).  Bad values within those not-taken branches must not cause exceptions (test case L289).
1. The `isnan` check is different than `if_then_else`.  In the latter, the `ExpressionNaN` is within a dead sub-expression branch.  In the former, it appears in an evaluated trunk.  That goes against rule (1) where a `NaN` anywhere in a computation (other than dead code) is an error.

I think the general premise is that if NaN can arise from a domain error or similar, then the correct formulation is to guard the computation using `if_then_else` specifying the valid domain of the input expression -- not to compute anyway than then check `isnan` after the fact.

So we should write `if_then_else(x >= 0, sqrt(x), 0)` not `if_then_else(isnan(sqrt(x)), 0, sqrt(x))`.

jwnimmer-tri, 2020-03-23:
Or perhaps more to the point: I suspect that neither @soonho-tri nor myself foresaw using NAN across and API boundary to indicate presence vs absence.

I'll open the integrator issue later today.

soonho-tri, 2020-03-23:
> we should provide isnan(Expression) for the common case of non-NaN value returning False.

+1 for this. I'll check things and open a PR.

jwnimmer-tri, 2020-03-23:
I think test case L279 proves that we already have that?

soonho-tri, 2020-03-23:
@jwnimmer-tri , right. We already have it:

```c++
 bool FormulaIsnan::Evaluate(const Environment& env) const {
   // Note that it throws std::runtime_error if it detects NaN during evaluation.
   return std::isnan(e_.Evaluate(env));
 }
```

If `static_cast<bool>(isnan(Expression::NaN()))` is a real blocker. How about changing `FormulaIsnan::Evaluate` to check the existence of `NaN` in its subtrees and return `true` if found?
Note: because of possible existence of `if-then-else` in its sub-expressions, the `NaN` check should involve evaluation of conditional expressions in ITE, which can throw an exception when `NaN` is computed.

jwnimmer-tri, 2020-03-23:
For the reasons outlined above and in #12904, I think we should stop using NaN as a gambler's `optional<double>`, instead of trying to add complexity to Expression / Formula that allow the programmer to win the bet sometimes.

sherm1, 2020-03-23:
`optional` seems like a much better idea than NaN for the integrators!

edrumwri, 2020-03-23:
Since you guys (mainly Nimmer) are repeatedly maligning my use of NaN's to signal a different mode of operation, allow me to point out that we didn't have access to optional when I wrote the integrators!

jwnimmer-tri, 2020-03-24:
> I agree that the pattern with NaNs in IntegratorBase is not ideal. @jwnimmer-tri -- can you open an issue for it?

=> #12927

jwnimmer-tri, 2020-03-24:
Perhaps the remaining action to resolve this issue is to improve Expression's documentation?

I've been wanting for a while to have a "symbolic expression" group page in Doxygen to tie it all together.  We could explain the design for NaNs there, perhaps.

RussTedrake, 2020-03-24:
that's great.  at very least, the docstring for `symbolic::isnan` could explain the reasoning.  It is correct as it is, but was not enough for me to understand.

soonho-tri, 2020-08-09:
Closed by #12938.

closed 2020-08-09 by soonho-tri