RobotLocomotion/drake #12926: pydrake: Bad behavior when passing `[None]` to C++ site accepting `VectorXd`

EricCousineau-TRI, 2020-03-23:
Per this Slack convo: https://drakedevelopers.slack.com/archives/C2CK4CWE7/p1584994164030500

@mpetersen94 found that this simulation would hang, even with `AdvanceTo(0.)`:
https://gist.github.com/mpetersen94/28b5f1a67347333afa3ce98e81e73054
@jwnimmer-tri pointed out that `set_tau` was return `None`, but was somehow eating the error.

Here's a min repro (basing off of 458f95c):
```py
source = ConstantVectorSource([None])
context = source.CreateDefaultContext()
y = source.get_output_port(0).Eval(context)
print(y)  # Prints: [ nan]
```

EricCousineau-TRI, 2020-03-23:
More minimal:
[branch](https://github.com/EricCousineau-TRI/drake/tree/feature-2020-03-23-py-ub) | [commit](https://github.com/EricCousineau-TRI/drake/commit/c20ae9d3005964fc02ca4a29e43e218dad7b34c2)

EricCousineau-TRI, 2020-03-23:
Also a problem in upstream, checked in Drake via 3ad30bdd4

jwnimmer-tri, 2020-03-23:
I'm not displeased with the None => NaN conversion.  To me, the more important question is why the simulator is allowing the NaN to cause it to infinite loop.

EricCousineau-TRI, 2020-03-23:
Aye. Well, the None => NaN is actually part of NumPy, so: d'oh!
https://github.com/pybind/pybind11/issues/2141#issuecomment-602903192

Will file separate issue about NaN-ness.

closed 2020-03-23 by EricCousineau-TRI

sherm1, 2020-03-23:
See #12647.

EricCousineau-TRI, 2020-03-23:
Thanks! Cross-posted there.