RobotLocomotion/drake #12927: Integrators should not use NaN to mean "unset"

jwnimmer-tri, 2020-03-24:
(Inspired by #12923.)

Currently, `IntegratorBase` and its derived classes use NaN to indicate "unset" or "not present" for some scalar-valued quantities, e.g., `max_step_size`.  This makes the code more difficult to read and reason about because then things like `h < max_step` and `h >= max_step` are not complementary -- they are both `false`.  For example, `if (max_step_size_ < req_min_step_size_)` is used to check that the two values are sensible relative to each other, but if one of them is NaN, the check has no effect.

We should refactor the integrators to avoid purposeful NaNs -- perhaps by using `optional<double>` or other means.

Note that even though this issue was filed at almost the same time as #12647, they are not really related.  That issue is about state-derived computations by a System returning NaN.  This issue is about values that are under the control of the integrator being set to NaN explicitly.

antequ, 2020-04-04:
I agree with the sentiments expressed here; it causes a lot of developer burden and our current code is pretty brittle. 

At the same time, we might also want to consider adding some tests just to ensure our behavior around NaNs and other special floating point values are expected and don't change as we modify integrator code in the future. I'm not sure specifically how we would add these, as there are so many scenarios.

sherm1, 2020-04-04:
Also see #12647 on the topic of integrators and NaNs.

jwnimmer-tri, 2020-04-06:
Some methods are currently called `get/set_maximum_step_size` but everywhere else in the code calls it `max_step_size` (not `maximum_step_size` spelled out).  I suggest deprecating the old longer method names concurrently with changing their nan-semantics.