RobotLocomotion/drake #12942: Unify treatment of forces in MultibodyPlant

sherm1, 2020-03-26:
### Background

In PR #12895 I wanted to do some simple bookkeeping of energy and power in MultibodyPlant. My thought was to sum over the ForceElement's energy and power methods. However, it turns out that ForceElements are by no means the only way forces get applied to a MBPlant. There are in fact a bunch of essentially unrelated ways forces get applied, and there is no uniform way to accumulate power contributions from them. In addition, as complained about in #12911 force introspection is not currently possible and the ad hoc treatment of different force sources makes that difficult to fix.

Here is an attempt to enumerate the current collection of forces, based on code archeology:
- ForceElements (yay!)
- Joint dampers (only available for 1-dof joints)
- Joint limits (only for 1-dof joints in discrete mode, not clear why discrete only)
- Joint actuation forces (produced by JointActuators which are not currently ForceElements)
- Applied external spatial forces (from an input port)
- Applied generalized forces (from an input port)
- Contact forces (different between discrete and continuous plants)
- Have I missed any?

### Proposal

Force should be applied to a MultibodyPlant only by ForceElements. We should create ForceElements capable of applying each of our internally-generated forces, and create those as needed. A good example is provided by UniformGravityFieldElement which is automatically created on construction. There are already TODOs in the code stating that joint damping should have been handled via ForceElement; we should also do that for actuators, joint limit forces, applied body forces, the generalized forces input port, contact forces, and likely a bunch more to come like joint friction forces.

After that, forces, energy, and power would be treated uniformly by iterating through the ForceElements and accumulating the appropriate quantity with no ad hoc code needed. Introspection would be simplified since all possible forces could be found in the collection of ForceElements.

amcastro-tri, 2020-03-26:
Thank you for the detailed summary @sherm1. 
Interesting, do you think that input port applied forces should also be `ForceElement`s?

amcastro-tri, 2020-03-26:
Transcribing from comment in the review of #12895. 

For power I believe that MBP could have another method like `MBP::CalcExternallyAppliedWork()`? which would only consider the work applied by input ports. It'd be a clean concept.

sherm1, 2020-03-26:
> Interesting, do you think that input port applied forces should also be ForceElements?

Yes, my current proposal is that _all_ forces that affect an MBP should run through ForceElement so that we can localize all the energy and power computations there. It would be easy to make an "applied generalized forces" force element -- essentially it would just read the corresponding input port, add the result to the generalized forces, dot with the generalized velocities to generate non-conservative power.

If that's too weird I wouldn't object to separating internally-generated forces from externally-applied ones, but I think we should first discuss why we can't have ForceElements be the sole source of forces.

Another conversation worth having is to decide who should handle the bookkeeping for forces that are applied to a System via its input ports. I _think_ the right answer is that the destination System should do the bookkeeping, since stored vs kinetic energy depend on the destination's internals, and power depends on the destination's velocities. A simple example would be a Diagram consisting of a one-actuator MultibodyPlant and a ConstantValueSource plugged into its actuator port. The source system can't be asked to bookkeep the power applied by the actuator since it has no idea to what use its output is being put.

amcastro-tri, 2020-03-27:
> Yes, my current proposal is that all forces that affect an MBP should run through ForceElement 

ok.

>  I think the right answer is that the destination System should do the bookkeeping

also ok.

However, I don't think a good idea to place sources of energy in `CalcNonConservativePower()`  just because we don't have a better place. That's why I suggested `MBP::CalcExternallyAppliedPower()`.

Then your energy budget would be split as:
- Actually stored energy:
  - CalcKineticEnergy()
  - CalcPotentialEnergy()
- Internal power:
  - CalcConservativePower()
  - CalcNonConservativePower()
- External sources:
  - CalcExternallyAppliedPower() (no objection if through force elements).

sherm1, 2020-03-27:
Makes sense to me to split out external power in MBP. We would probably have to change some or all of the MBP method names to avoid confusion with the System virtuals. There is no equivalent of externally-applied power at the top-level Diagram since all forces are internal at that level. That is, the System framework method `CalcNonConservativePower()` _must_ include what MBP thinks is its externally-applied power. So the meanings of kinetic/potential/conservative/nonconservative are defined already at the System level and can't be changed by MBP. However, we can have:
- MBP::CalcInternalNonConservativePower()
- MBP::CalcExternallyAppliedNonConservativePower()
- Then CalcNonConservativePower() override is the sum of those two.

BTW see #12752 for related upcoming changes to the System API.