RobotLocomotion/drake #12947: API to retrieve different spatial inertias

amcastro-tri, 2020-03-26:
It'd be nice having an API to retrieve different `SpatialInertia`s in their own Doxygen group. It'd include:
- `EvalBodySpatialInertiaInWorld()`, `M_BBo_W`
- `EvalCompositeBodySpatialInertiaInWorld()`, `R_BBo_W`
- `EvalArticulatedBodySpatialInertiaInWorld()`, `P_BBo_W` (actually an `ArticulatedBodyInertia`).

amcastro-tri, 2020-03-26:
Notice we do have `MultibodyTreeSystem::EvalSpatialInertiaInWorldCache()` but this, and others, do not show up in MBP's Doxygen.

sherm1, 2020-03-26:
These would be good to have. The ones that return cache entries shouldn't be included though because those are `internal::`.

Not clear what "ArticulatedBodySpatialInertia" means above? A spatial inertia (10 numbers) is too restrictive to hold an articulated body inertia (21 numbers).

amcastro-tri, 2020-03-26:
> Not clear what "ArticulatedBodySpatialInertia" means above? 

good catch, the result of copy/paste. Thanks.