RobotLocomotion/drake #12962: opengl_context_test error reports under valgrind

jwnimmer-tri, 2020-03-28:
``//geometry/render/gl_renderer:opengl_context_test` has lots of memcheck (valgrind) complains.

More details to follow.

jwnimmer-tri, 2020-03-30:
Now that I've updated my write-up, over to @SeanCurtis-TRI for triage.

tehbelinda, 2020-03-30:
Just adding some notes:
I see the same issue with CI when using Mesa. I also have errors, but less errors, when using Nvidia
```
belindateh@T480-BTeh:~/drake$ c-bazel test -c dbg --config=fastmemcheck //geometry/render/gl_renderer:opengl_context_test
INFO: Analyzed target //geometry/render/gl_renderer:opengl_context_test (0 packages loaded, 0 targets configured).
INFO: Found 1 test target...
FAIL: //geometry/render/gl_renderer:opengl_context_test (see /home/belindateh/.cache/bazel/_bazel_belindateh/850f926919c2ff48c6e09dc437003697/execroot/drake/bazel-out/k8-dbg/testlogs/geometry/render/gl_renderer/opengl_context_test/test.log)
INFO: From Testing //geometry/render/gl_renderer:opengl_context_test:
==================== Test output for //geometry/render/gl_renderer:opengl_context_test:
==17== Memcheck, a memory error detector
==17== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==17== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==17== Command: geometry/render/gl_renderer/opengl_context_test
==17== 
Using drake_cc_googletest_main.cc

[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from OpenGlContext
[ RUN      ] OpenGlContext.GetContext
[2020-03-30 18:44:19.523] [console] [info] Vendor: NVIDIA Corporation
[       OK ] OpenGlContext.GetContext (4804 ms)
[----------] 1 test from OpenGlContext (4823 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (4891 ms total)
[  PASSED  ] 1 test.
==17== 
==17== HEAP SUMMARY:
==17==     in use at exit: 149,789 bytes in 566 blocks
==17==   total heap usage: 5,422 allocs, 4,856 frees, 196,669,525 bytes allocated
==17== 
==17== 128 bytes in 1 blocks are definitely lost in loss record 206 of 266
==17==    at 0x4034B35: calloc (vg_replace_malloc.c:752)
==17==    by 0x8C2E995: ??? (in /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21)
==17==    by 0x8C22E38: ??? (in /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21)
==17==    by 0x8C227C1: ??? (in /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21)
==17==    by 0x8C32247: ??? (in /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21)
==17==    by 0x79235B5: ??? (in /usr/lib/x86_64-linux-gnu/libGLX_nvidia.so.435.21)
==17==    by 0x40106EB: call_init (dl-init.c:58)
==17==    by 0x40106EB: _dl_init (dl-init.c:119)
==17==    by 0x40151FE: dl_open_worker (dl-open.c:522)
==17==    by 0x61132DE: _dl_catch_exception (dl-error-skeleton.c:196)
==17==    by 0x40147C9: _dl_open (dl-open.c:605)
==17==    by 0x6AACF95: dlopen_doit (dlopen.c:66)
==17==    by 0x61132DE: _dl_catch_exception (dl-error-skeleton.c:196)
==17==    by 0x611336E: _dl_catch_error (dl-error-skeleton.c:215)
==17==    by 0x6AAD734: _dlerror_run (dlerror.c:162)
==17==    by 0x6AAD050: dlopen@@GLIBC_2.2.5 (dlopen.c:87)
==17==    by 0x63A5605: ??? (in /usr/lib/x86_64-linux-gnu/libGLX.so.0.0.0)
==17== 
{
   <insert_a_suppression_name_here>
   Memcheck:Leak
   match-leak-kinds: definite
   fun:calloc
   obj:/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21
   obj:/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21
   obj:/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21
   obj:/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.435.21
   obj:/usr/lib/x86_64-linux-gnu/libGLX_nvidia.so.435.21
   fun:call_init
   fun:_dl_init
   fun:dl_open_worker
   fun:_dl_catch_exception
   fun:_dl_open
   fun:dlopen_doit
   fun:_dl_catch_exception
   fun:_dl_catch_error
   fun:_dlerror_run
[... 115 lines cut ...]
   fun:_dl_init
   fun:dl_open_worker
   fun:_dl_catch_exception
   fun:_dl_open
   fun:dlopen_doit
   fun:_dl_catch_exception
   fun:_dl_catch_error
   fun:_dlerror_run
   fun:dlopen@@GLIBC_2.2.5
   obj:/usr/lib/x86_64-linux-gnu/libGLX.so.0.0.0
}
==17== LEAK SUMMARY:
==17==    definitely lost: 6,792 bytes in 4 blocks
==17==    indirectly lost: 76,060 bytes in 409 blocks
==17==      possibly lost: 0 bytes in 0 blocks
==17==    still reachable: 66,937 bytes in 153 blocks
==17==         suppressed: 0 bytes in 0 blocks
==17== Reachable blocks (those to which a pointer was found) are not shown.
==17== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==17== 
==17== For counts of detected and suppressed errors, rerun with: -v
==17== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
================================================================================

```

tehbelinda, 2020-03-30:
Potentially related? https://github.com/RobotLocomotion/drake/issues/7520
The vtk_test_tags which we use for the render engines already have the no_memcheck tag and link to #7520

SeanCurtis-TRI, 2020-03-31:
Certainly the fact that the stack trace in #7520 includes `libX11.so.6.3.0` is interesting.  And the `swrast_dri.so` which appears there and in the mesa-version is also telling. (One would presume that `swrast_dri` is some form of software rasterizer.

jwnimmer-tri, 2023-05-23:
With some experience under our belt, I think our current posture is that trying to run GL drivers under a memory checker is too much effort for too little of a benefit.  Closing as "won't fix".

closed 2023-05-23 by jwnimmer-tri