RobotLocomotion/drake #12965: Real-time state advancer

edrumwri, 2020-03-28:
We've been developing control and state estimation systems using the systems framework (i.e., by deriving systems from LeafSystem).

Though I'm sure we have yet to encounter all possible challenges to producing deterministic computation times, the framework has been working well in this regard thus far. Heap allocations can generally be avoided without much effort.

One of the mechanisms that was not designed for real-time operation is Simulator. We would like to advance the state of our systems alongside wall-clock-time in a real-time-safe manner (i.e., by ensuring that computation times are deterministic).

We've been thinking of building a real-time state advancement mechanism for this purpose and PR'ing it for inclusion into Drake, so is there interest from the Drake developers and larger Drake community in us doing so? And, does anyone with advanced knowledge of the systems framework (e.g., @jwnimmer-tri, @sherm1) anticipate any especially difficult issues we would encounter going down this road?

sherm1, 2020-03-28:
I don't have a clear picture yet of how this would differ from the regular Simulator using a fixed-step integrator and real time throttling. Can you say more about the differences?

edrumwri, 2020-03-30:
Sure, though I won't claim clarity on the entire picture yet because we're very early in the design process.

Here are the differences that I see (also incorporating the two that @sherm1 mentioned)) of the real-time state advancer compared to Simulator:
* Time would advance in lockstep with wall-time (i.e., no need for, e.g., get_actual_real_time_rate() and the like)  
* Only integrators with predictable computations would be admissible- no implicit integrators, no error controlled integrators
* Witness functions trigger times would not be isolated (debatable, might be possible to do this but using number of integration steps rather than using an isolation tolerance)
* No publishing would be performed (debatable) since publishing tends to result in I/O (with attending nondeterministic computation times)
* (Internal) No heap allocations during HandleXUpdate()

jwnimmer-tri, 2020-03-30:
To me, those sound more like tweaks to the current Simulator than something that justifies a new implementation.

The "don't choose the wrong integrator (2)" and "don't declare expensive publish events (4)" and "don't set the realtime rate to other than 1.0 (1)" are only preventing user mistakes.  The possibility to use a Simulator under those conditions already exists today.

I think we could meet the "make it simple to prevent mistakes" goal with a Simulator-creating factory function that conveniently configures all of those properties in the recommended manner.

Adding new settings to establish an upper bound on the computation performed for witness isolation (3) sounds useful, even for non-realtime simulation.  That could be a new tuning knob on the current Simulator?

We could look into heap use during HandleXUpdate (5).  We have the ability to unit test for this.

edrumwri, 2020-03-30:
> To me, those sound more like tweaks to the current Simulator than something that justifies a new implementation.

Your proposal would suit my needs and would permit ignoring the problem of how to reuse code between the two mechanisms (Simulator and RTSA). So thumbs up from me.

But how could you address the problem that a Simulator is not the obvious name of a tool for accomplishing this task? A Simulator simulates a virtual version of a system while users would be looking for some way to advance the state for use by some real robot/system. Not saying that this isn't addressable, just curious how you would plan to direct people to using this tool for this purpose.

sherm1, 2020-03-30:
If the name is the only problem I think that just requires documentation. The Drake Simulator is the solver that advances a Drake System through time -- that's still happening here I assume? It's just that the Drake System diagram might not include a simulated robot?

edrumwri, 2020-03-31:
Yes, the Simulator would still advance the system through time, but I disagree with the characterization that "it's just that the [diagram] might not include a simulated robot".  It's really that the Simulator would be co-opted to a new purpose, namely advancing the state in concert with wall-time. Some of the Simulator concepts, like "throttling the simulation" would no longer be a natural fit, at least in my mind, because the Simulator would no longer be the mechanism that actually advances time: nature replaces it as that mechanism.

The distinctions are only to say that if I knew little about Drake (and Simulink†), _Simulator would be the last place I would look to provide this capability_: when the name of something does not indicate a second, very different use than its nominal one (to reiterate, I wouldn't expect a Simulator to be the tool to advance the state of my real-time control+state estimation diagram through time), documentation must fight an uphill battle.

† Simulink doesn't do any better IMO: it provides an option called "Simulation mode" = "external" to advance the state through time on real hardware.

jwnimmer-tri, 2020-03-31:
When we think of a good name, we can use that name for the factory method that creates and returns a Simulator instance configured for this purpose.  (As well as create appropriate documentation sections or groups, or a new source folder, examples, or etc. that point people to it.)

ggould-tri, 2020-03-31:
A version of this question came up in Drake "system 2" design four years ago.  The objections I had then do not bite so hard today (particularly as we have dropped Windows and MEX support) but I figured I should scribe them here for posterity.

* Realtime control is tightly coupled to system safety.  To say that Drake is suitable for realtime control requires either a strong disclaimer about safety-critical use or else a layer in the framework for positive control interlocking and all of that other stuff that it is irresponsible to touch safety systems without.
* Any realtime control design must, at more or less step zero, have well-defined fallbacks for when a component fails to meet realtime guarantees or acts in a way that will prevent other components from meeting theirs.  This will involve OS-level primitives and will not be cross-platform compatible.
* Certain features of Drake are intrinsically unsuited to realtime control (eg, anything that touches SNOPT) and we would do well to wall these off in some way from any system that commits to perform at realtime.
* Drake diagrams have pull logic and other "action at a distance" properties that make it extremely hard to say that a particular computation in one module will not be invoked by private mechanisms of of a different, distant module (eg in forcing the calculation of an output port).
  * Moreover modern Drake mitigates this effect through the use of caching, which is notorious for causing a program to appear to meet its realtime goals in sunny-day conditions but betray you in practice.  It is notable that many safety system design guidelines I have met implicitly or explicitly forbid caching.

None of this is to say that we shouldn't do this!  Drake absolutely should be able to do interesting control things!  But we should spare a few minutes' thought for how Drake's first google result will never be the five-whys investigation of some future safety incident by a car's steering controller.

edrumwri, 2020-03-31:
Excellent points, Grant. Thank you for dredging these up.

sherm1, 2020-03-31:
> It's really that the Simulator would be co-opted to a new purpose, namely advancing the state in concert with wall-time.

I don't understand how this is a difference in practice. Suppose I decide to advance the system by 1ms of "wall clock time" from t to t+1ms. If that step takes half a ms of compute time, the Simulator arrives at the end of its step at t+0.5ms (wall clock). Then it has to do nothing for 0.5ms to allow wall clock to catch up. That's exactly what happens now with real time factor=1. Can you explain how your proposed use is different?

edrumwri, 2020-03-31:
> I don't understand how this is a difference in practice. Suppose I decide to advance the system by 1ms of "wall clock time" from t to t+1ms. If that step takes half a ms of compute time, the [...]

Since you broached it, here's a difference where a real-time state advancer should probably do things differently than Simulator: Simulator uses sleep_until(x) in PauseIfTooFast(.), where x is the desired real-time target.  Some slop there is probably fine for non-real-time applications. For real-time applications, oversleeping the allocated interval is not acceptable, which would seem to indicate some other strategy (perhaps thread::yield(), which would be a less-good choice for Simulator).

sherm1, 2020-03-31:
A more-precise PauseIfTooFast() option could be added to the existing Simulator but nothing will make it a hard real-time system along the lines Grant presented above. For that we'd need a real time OS with guaranteed response times -- I don't think Ubuntu can provide that. OTOH, if you don't need the full hard RT treatment, then I think the issues are empirical -- how much slop can you tolerate, and how close can we bring the existing Simulator to that?

edrumwri, 2020-03-31:
You're right- it's infeasible to make Ubuntu hard real time, and that's okay. We'd like to be able to minimize/eliminate sources of running time nondeterminism when we advance our diagram's state, so if it's to be Simulator that's advancing State, a few things will likely have to be modified to better support real-time operation. It's not clear to me that any of the Simulator modifications to this effect should be empirical; I don't see why one would use sleep_for(.) in real-time operation or yield() in non-real-time operation, for example.

I still agree with the suggestion from @jwnimmer-tri that the best path to get what we want is through modifying Simulator.

jwnimmer-tri, 2020-03-31:
TBH if I were the application engineer, I'd never rely on AdvanceTo's self-interlocking anyway.  I'd set its realtime rate to 0.0 (no sleeping) and manually tick it with the exact step size I wanted, while using realtime software watchdogs in my calling code to fail fast (as Grant so well notes) when we exceed our budget, as well as monitoring how close to it we're coming.  I don't think building the watchdogs and monitoring into Drake is a good idea because they are always very application specific and related to the user's own safety case, so frankly I don't think requirement (1) should even really be part of the library design.  It should be fully implemented on the end-user side -- there is not really any valid way for our library to anticipate users' needs here.

edrumwri, 2020-03-31:
> I don't think requirement (1) should even really be part of the library design. It should be fully implemented on the end-user side -- there is not really any valid way for our library to anticipate users' needs here.

That is indeed one of options we've considered. If this feature weren't built into Drake, it wouldn't be the end of the world for us, though it's not obvious to me that some provisions couldn't be made in Drake to facilitate this operation. I realize that the breadth of Drake's mission has meant that its use for control of real-time systems has not been baked-in from the beginning. Still, it seems a crime to have access to this well designed framework for developing controllers and state estimators but have to struggle to apply it to real systems!

sherm1, 2020-03-31:
A more nuts-and-bolts question: how much better is jitter with a thread::yield() busy-wait vs. sleep_until()? yield() still seems like it could be highly variable if anything else is going on. Maybe just making the Simulator process higher priority than anything else would make sleep_until() respond quickly?

jwnimmer-tri, 2020-04-01:
There are many nuances of scheduling and wake-ups (see [sched(7)](http://man7.org/linux/man-pages/man7/sched.7.html) for starters), some of which is touched on in our [kuka_driver](https://github.com/RobotLocomotion/drake-iiwa-driver/blob/master/kuka-driver/kuka_driver.cc).  I don't think the simulator should wade too deeply into those waters at this point.  I'd rather us just focus on the heap allocations for now.

As TRI publishes more of our drivers and controllers, users will have more examples of time-sensitive control loops and different ways to approach them.  As long as the simulator doesn't get in the way of users composing their own solution, I think that will suit for a while.  Even `std::chrono` is known to be sub-optimal for many use cases -- good clocks are difficult to come by, and are usually platform-specific (maybe C++23 will change that?).

It's also the case in my experience that a software clock won't usually be the time master.  Usually it would be the clock on the low level status+control hardware that is sending notionally-periodic status updates, to which the "simulator" must quickly process and respond to.  The cpu clock of the "simulator" is used as a fallback / watchdog, and does not supplant a hardware clock.  (At least, that's how I've always done it.)

All of which is to say -- I'd advocate that we focus on the things that users _can't_ do on their own.  That would be things like drake-internal heap allocations within inner loops, and the engineering approach to reason about sunny day testing also covering rainy day codepaths.

edrumwri, 2020-04-01:
> All of which is to say -- I'd advocate that we focus on the things that users _can't_ do on their own. That would be things like drake-internal heap allocations within inner loops, and the engineering approach to reason about sunny day testing also covering rainy day codepaths.

SGTM!

closed 2024-03-20 by jwnimmer-tri