RobotLocomotion/drake #12989: Homebrew cellar paths can change out from under us

jwnimmer-tri, 2020-04-02:
I have a macOS build that I visit from time to time to debug macOS-specific problems.  On my latest visit there, I got this error during a build:
```console
ERROR: .../external/dreal/dreal/contractor/BUILD.bazel:14:1: Couldn't build file external/dreal/dreal/contractor/_objs/contractor/contractor_seq.pic.o: C++ compilation of rule '@dreal//dreal/contractor:contractor' failed (Exit 1) wrapped_clang failed: error executing command external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG ... (remaining 106 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from external/dreal/dreal/contractor/contractor_seq.cc:1:
In file included from external/dreal/dreal/contractor/contractor_seq.h:6:
In file included from external/dreal/dreal/contractor/contractor.h:6:
In file included from external/ibex/include/_usr_local_opt_ibex@2.7.4_include/./ibex.h:15:
In file included from external/ibex/include/_usr_local_opt_ibex@2.7.4_include_ibex/ibex_LinearizerAffine2.h:16:
In file included from external/ibex/include/_usr_local_opt_ibex@2.7.4_include_ibex/ibex_Linearizer.h:14:
external/ibex/include/_usr_local_opt_ibex@2.7.4_include_ibex/ibex_LPSolver.h:28:10: fatal error: 'ClpSimplex.hpp' file not found
#include "ClpSimplex.hpp"
         ^~~~~~~~~~~~~~~~
1 error generated.
```

The problem seems to be that the include path for clp changed during a homebrew upgrade since the last time I'd built, but nothing in bazel knew to re-run pkg-config to obtain the new path.

In #12988, I've added support for `configure = True` so that a "bazel sync --configure" (or full "bazel sync") would recover from this.  But the user might not know to run that command.

@jamiesnape suggests that our pkg-config rules should learn to rewrite `/usr/local/Cellar/FORMULA/VERSION` to `/usr/local/opt/FORMULA` as a mitigation.  I agree.

I wonder if we should also run "bazel sync --configure" or even a full "bazel sync" as a late step during "source_distribution/install_prereqs.sh" so that the usual advice of "if your build fails on external code, re-run install prereqs to repair it" invariant holds.

soonho-tri, 2020-04-02:
> I wonder if we should also run "bazel sync --configure" or even a full "bazel sync" as a late step during "source_distribution/install_prereqs.sh" so that the usual advice of "if your build fails on external code, re-run install prereqs to repair it" invariant holds.

I liked this option. @jamiesnape , is there any harm in running `bazel sync`?

EricCousineau-TRI, 2020-04-02:
(Keeping Jamie only on assignee list)

jamiesnape, 2020-04-02:
> I liked this option. @jamiesnape , is there any harm in running bazel sync?

Philosophically, it seems a little beyond the scope of the file, but it would not hurt so long as you checked the script was actually run from within a workspace (I often download just the raw scripts when provisioning, for example). Not sure if there is a clean way to detect if anything was actually upgraded and display a warning, but that would be my preference if it were possible.

jwnimmer-tri, 2020-04-03:
I agree with that sentiment.

For now the better first step sounds like rewriting homebrew paths in particular.

jwnimmer-tri, 2023-01-27:
We are using Homebrew for fewer and fewer libraries these days.  Closing this as "won't fix" for now.

If we find that some library like `glib` or something is still hurting us, we can reopen.

closed 2023-01-27 by jwnimmer-tri