RobotLocomotion/drake #13000: [2020-04-04 08:06:00.825] [console] [warning] TypeHash<T> cannot operate on T=drake::multibody::MultibodyPlant<double>::HydroelasticFallbackCacheData; Value<T> may suffer from slightly impaired performance.

54WC, 2020-04-04:
This warning appeared on bash console since last Drake/Underactuated python update. ( Follow Course Textbook for 2020 Underactuated Robotics: A3.1 Install Drake and A3.2 Download the textbook supplement) 
The warning message show on the console can't help me to remove warning.  
Could you advice me how to fix it ? Is it related to my code , console setup or Drake source code? 
Thank you. 

[2020-04-04 08:06:00.825] [console] [warning] TypeHash<T> cannot operate on T=drake::multibody::MultibodyPlant<double>::HydroelasticFallbackCacheData; Value<T> may suffer from slightly impaired performance. If T uses a single non-type template parameter, adding 'using NonTypeTemplateParameter = ...;' will enable hashing. See drake/common/test/value_test.cc for an example. This is the first instance of an impaired T within this process. Additional instances will not be warned about, but you may set the drake::log() level to 'debug' to see all instances.

SeanCurtis-TRI, 2020-04-04:
The most important thing I can tell you is that you can freely ignore the warning. There's nothing you did to cause and basically nothing you can do to remove it.

The second most important thing is that you're only getting this message because the build of Drake you're currently using is mildly out of date. This issue (and a related issue) were resolved in #12953 and #12932, seven and eleven days ago, respectively.

So, until things are updated, just ignore it.  :)

SeanCurtis-TRI, 2020-04-04:
I'm going to close this because of the explanation given (solved in master, nothing to do or worry about if you're older than master). Feel free to re-open if this is still causing you problems.

closed 2020-04-04 by SeanCurtis-TRI