RobotLocomotion/drake #13021: Need parsing support for the bushing force element

sherm1, 2020-04-08:
The LinearBushingRollPitchYaw element currently can only be defined through the C++ API. Should be possible to add it explicitly from sdf or udf via a custom `drake:` tag. 

In addition we may want to consider creating a bushing automatically when we detect a topological loop. That's a much bigger project and NOT the topic of this issue! I've captured Russ's comment about it below for future reference.

Assigning initially to @joemasterjohn who is currently working on an example that has a bushing in it.
/cc @amcastro-tri @mitiguy

From discussion in #11831:
@jwnimmer-tri: I would like to see parser support for this (`<drake:linear_bushing_roll_pitch_yaw>`) before calling the story complete. I don't see an issue for that yet? Even in Anzu we're substantially less likely to use it without parser support.

@RussTedrake: I was thinking that it would not be a new tag, but that it would be added automatically if the sdf contained a closed kinematic chain. Had a conversation with @joemasterjohn along those lines.

jwnimmer-tri, 2020-04-09:
I realized later -- we'll need a new tag for sure, because we have plans to use the bushing other than for kinematic chains.  (Is that correct, @ggould-tri?)

ggould-tri, 2020-04-09:
My plan to use it for eg our rack joints in anzu is predicated on it ending up in sdf, yes.

That is not necessarily a high priority item for your design -- anzu isn't going to block on this in any near future scenario -- but I would argue that if sdf is meant to be the primary way that scenario data gets into Drake then all Drake mechanical features should be accessible via sdf.  The alternative means of scenario assembly are not user friendly after all.

sherm1, 2020-04-09:
Agreed -- there are many possible uses for a bushing besides loop closure.

sherm1, 2020-05-26:
@joemasterjohn should this issue be closed now?

joemasterjohn, 2020-05-27:
Yes, this issue is resolved by #13255.

closed 2020-05-27 by joemasterjohn