RobotLocomotion/drake #13024: Mock LCM does not unsubscribe on delete

edrumwri, 2020-04-09:
Per this line, https://github.com/RobotLocomotion/drake/blob/8f449959b0bd4acf8495d1a0a1dd4013da0d7be1/lcm/drake_mock_lcm.cc#L33, the Mock LCM does not unsubscribe on delete. This caused a segfault in our code when a subscription that should have been deleted from the interface was instead handled.

ggould-tri, 2020-04-09:
It's not obvious to me which team should own this issue; I'm assigning it to dynamics but feel free to reassign it if you have a better idea.

SeanCurtis-TRI, 2020-04-09:
For my own clarification -- how did a method on the deleted LCM interface get invoked? The mock interface uniquely owns its subscriptions (and the handlers). Those handlers are only exercised in `DrakeMockLcm::HandleSubscriptions()`, so to invoke one of those subscription handlers `HandleSubscriptions()` must have been invoked on a deleted mock interface?  What am I missing?

edrumwri, 2020-04-09:
> For my own clarification -- how did a method on the deleted LCM interface get invoked? The mock interface uniquely owns its subscriptions (and the handlers). Those handlers are only exercised in [...]

It wasn't invoked on a deleted mock interface. It was invoked on a deleted functional.

edrumwri, 2020-04-09:
Good question. I misdiagnosed the problem, and I can't say too much more about it. This issue is no longer relevant to our problem, but I believe it should stay open since the mock interface does not do what it promises.

jwnimmer-tri, 2020-04-09:
It would be fine to add support for unsubscribe, but to be clear -- unsubscription is optional in the API contract, so even as it stands today all promises are met.

https://github.com/RobotLocomotion/drake/blob/92d8f2dc690e969a5d2ac83c2774d5d994d5c9d0/lcm/drake_lcm_interface.h#L78-L79

https://github.com/RobotLocomotion/drake/blob/92d8f2dc690e969a5d2ac83c2774d5d994d5c9d0/lcm/drake_lcm_interface.h#L188-L189

edrumwri, 2020-04-09:
Sorry for the initial misdirection in this issue, but my problem has arisen from differences between the two Drake LCM interfaces. They're typically used very differently. DrakeLcm communicates just fine through a number of DrakeLcm objects while DrakeMockLcm requires all communications go through a single object. Thus the two are not interchangeable, which would be ideal- one could be a drop-in replacement for the other.

I would think that the two interfaces could behave consistently if the DrakeMockLcm messages were stored statically. Is there a good reason that was not done?

jwnimmer-tri, 2020-04-09:
Are you aware of the `memq://` URL?  When passed to `DrakeLcm` constructor (or set via environment variable), I believe believe it does exactly what you want -- a process-local bus that does not touch the network.  That is the most drop-in replacement -- exactly the same code and control flow, just without ever hitting the network.

edrumwri, 2020-04-09:
> Are you aware of the `memq://` URL? When passed to `DrakeLcm` constructor (or set via environment variable), I believe believe it does exactly what you want -- a process-local bus that does not [...]

I'd seen mention of it looking through the LCM code, though I wasn't aware of its purpose (a grep in the LCM directory doesn't reveal much), and this approach still sounds nondeterministic, based on the docs in DrakeMockLcm: "It is similar to a DrakeLcm object with a `memq://` URL, but is guaranteed to behave deterministically (without a hidden background thread)".

jwnimmer-tri, 2020-04-09:
> ... this approach still sounds nondeterministic ...

The 100% determinism of DrakeMockLcm is helpful for narrow, focused unit tests.  The device(s) under test are given a pointer to it, and then the test author can can sequence the send and receive messages very previcely, without needing to add any retry-for-a-few-microseconds loops.

It's true that memq + DrakeLcm might require some small retry loops (though usually is fine), but in my experience in the cases where I've needed memq, I already have those retries in place.

> ... if the DrakeMockLcm messages were stored statically ...

The reasons DrakeMockLcm is not a singleton are (1) eww, singleton (2) action-at-a-distance is contrary to the goal of narrow, focused unit testing.  In cases where the application wants to use DrakeMockLsm as a singleton, the application can wrap it into a singleton pattern on its own in an application-specific way as appropriate.

I'm not keen to do that in Drake directly, because to me it's a best practice to pass the DrakeLcmInterface pointer into _every_ component that needs it, not to create multiple distinct DrakeLcm or DrakeMockLcm objects internal to classes or subsystems.  Each DrakeLcm object has associated costs (e.g., background thread, self-test, memory, cleanup, etc.) that should not be paid more than necessary.  Usually the top-level main() program (or a class that stands in for main) should allocate the DrakeLcm and then pass everywhere it's needed.  In that case, using one DrakeLcm vs one DrakeMockLcm hasn't been a difficult swap if we need to.

But mostly, I've been content with the `memq://` URL in the vast majority of circumstances.

In short, it would be fine to add support for unsubscribe, but I would not want to see Drake add a singleton.

edrumwri, 2020-04-09:
> In short, it would be fine to add support for unsubscribe, but I would not want to see Drake add a singleton.

That's fine, though I reiterate that I'm not a fan of the distinct operating modes (DrakeLcm vs. DrakeMockLcm). Since I'm not allowed to go into detail, I'll just say that our design decisions have required embedding the interfaces into classes. Fortunately, DrakeMockLcm is simple enough that I could copy the code into a proprietary class where I did indeed use a static variable. For our purposes, this solved my problem perfectly although having to maintain separate code is not a win.

jwnimmer-tri, 2020-04-09:
If you want singleton behavior then you can just wrap it in a singleton, no need to fork:
```c++
namespace myproject {
drake::lcm::DrakeMockLcm& global_lcm() {
  static drake::never_destroyed<drake::lcm::DrakeMockLcm> g_result;
  return g_result.access();
}
}  // namespace myproject
```

It is also still unclear to me why DrakeLcm + memq URL is unsuitable.  I don't think you'll be able to tell the difference between that and a singleton DrakeMockLcm.

jwnimmer-tri, 2020-04-09:
I went back to refresh my memory about how memq works.  I don't think there are any nondeterminism issues, actually.  I'll confirm for sure, but I suspect we can do away with DrakeMockLcm entirely.

sherm1, 2020-04-11:
@edrumwri does @jwnimmer-tri's proposal solve your problem, and if so can we close this?

PR #13028 (2020-04-11): lcm: Replace DrakeMockLcm implementation with memq
Closes #13024.

The `DrakeLcm` object with `memq://` backend works exactly like `DrakeMockLcm`. We've long since removed the extra sugar methods that used to live on `DrakeMockLcm` beyond those required by `DrakeLcmInterface`.

Remove all in-tree uses of `DrakeMockLcm`; it's clearer to write exactly what we mean without this synonym.  For tests it doesn't even matter, because we've set the default URL during bazel test:
https://github.com/RobotLocomotion/drake/blob/5a151dd5ff85a4674a3c2e5e177dfe0fe623d2ea/tools/bazel.rc#L40-L45

Perhaps we'll  [... cut]
closed 2020-04-11 by jwnimmer-tri