RobotLocomotion/drake #13026: pybind: Should ask CPython API maintainers to make gc reclamation be public API

EricCousineau-TRI, 2020-04-09:
Per convo with @jamiesnape, seems like Python 3.8 and 3.9 may start having some breakages.

I should make sure I make this request sooner than later (would have been better to have made it ~1 or 2 years ago):
https://github.com/RobotLocomotion/pybind11/blob/cac3edd971c70469aa710c13d59f9c29c7f4651f/include/pybind11/pytypes.h#L1465-L1475

\cc @ToffeeAlbina-TRI

EricCousineau-TRI, 2020-04-09:
For example, breakage in Cython for this very reason :(
https://bugs.python.org/issue35081#msg330045

EricCousineau-TRI, 2020-04-09:
Filed: https://bugs.python.org/issue40240

EricCousineau-TRI, 2020-04-09:
Ask made, so it may be in the works. For now, will close this, but should re-open once the internal API becomes some form of public and is controllable with a version macro.
(I will receive notifications on upstream issue.)

closed 2020-04-13 by jamiesnape

jwnimmer-tri, 2020-04-20:
FYI relates to the Drake 3.7 -> 3.8 transition from #13087.

PR #13027 (2020-04-13): workspace: Update pybind11 fork to latest commit (Python 3.8)
Resolves #13026

Requires:
- [x] Merge of https://github.com/RobotLocomotion/pybind11/pull/39
- [x] Merge of #13035

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13027)