RobotLocomotion/drake #13033: Compute UnitInertia from VolumeMesh

SeanCurtis-TRI, 2020-04-10:
Given the ubiquitous assumption of uniform density, there is no reason we shouldn't be able to compute the unit inertia for an arbitrary volume mesh and use that in dynamics.

SeanCurtis-TRI, 2020-04-10:
cc @amcastro-tri

jwnimmer-tri, 2020-04-10:
It would be neat to have this algorithm in master!  For now, I use meshlab to compute the inertia matrix and then `drake/manipulation/util/meshlab_to_sdf.py` to convert it to `*.sdf`'s XML format and commit the `*.sdf` file.  Evicting meshlab from the picture would make it even smoother.  I don't even need the runtime to do it automatically for me, I just need an obj,mass -> inertia subroutine.

SeanCurtis-TRI, 2020-04-10:
Well, we're a ways a way from obj -> volume mesh. We're currently not doing tetrahedralizations on arbitrary meshes -- that'll be a step in the future to directly enable "soft" meshes (a currently *unscheduled* step). But, by providing the functionality now, it'll slot in nicely then.

SeanCurtis-TRI, 2020-04-10:
Although, I strongly suspect unit inertia from surface mesh should be immediately doable as well...I'll confer with @mitiguy and correspondingly update the issue if it checks out.

SeanCurtis-TRI, 2020-04-10:
Yep; we're good. We should be able to produce the SpatialInertias for arbitrary meshes.

Reference for the implementer:

http://docsdrive.com/pdfs/sciencepublications/jmssp/2005/8-11.pdf

sherm1, 2020-04-10:
That reference looks good for a tet mesh. You mentioned doing this for surface meshes also?

SeanCurtis-TRI, 2020-04-10:
You use it for the surface mesh in the same way you can compute the volume of surface mesh: divergence theorem.

1. Pick reference point (origin, average vertex, whatever).
2. Compute the signed volume, centroid, and inertia tensor for every tet (some will be negative some positive) formed by a surface triangle and the reference point.
3. Combine all of the tets' quantities appropriately (noting that some will add to and some will remove from).

Done

SeanCurtis-TRI, 2020-04-10:
Essentially, the same way you create the inertia tensor for a thin spherical shell -- the tensor of the large volume minus the tensor of the small volume.

sherm1, 2020-04-10:
FYI regarding the tet inertia reference above: there are two sign conventions for products of inertia (the off-diagonals). Check with @mitiguy to make sure to use the Drake sign convention.

amcastro-tri, 2020-04-10:
I computed this in the past from the (water tight) surface of the volume. In that case you use Gauss's theorem to replace volume integrals for surface integrals.

amcastro-tri, 2020-04-10:
This document has all you need: https://www.geometrictools.com/Documentation/PolyhedralMassProperties.pdf

sherm1, 2020-04-10:
@jwnimmer-tri can you say more about the workflow you have in mind? It's easy for me to see how we could add another calculator to Drake's UnitInertia class. But I'm not seeing yet how that works in an sdf-oriented workflow. For example, here is Gazebo's recommended workflow: http://gazebosim.org/tutorials?tut=inertia&cat=build_robot

jwnimmer-tri, 2020-04-10:
@sherm1 that was exactly the tutorial we've been using, just semi-automated using https://github.com/RobotLocomotion/drake/blob/master/manipulation/util/meshlab_to_sdf.py.  My goal would be to update + rename that tool so that given an `*.obj` file from an artist (+ the `--mass_kg` command line argument), it creates the rough draft `*.sdf` exactly like that tool does now, but without the user needing to deal with meshlab.

I would then verify the file is valid, and commit it to git for code review, merge, etc. as usual.  It would _not_ be computing new inertias on the fly during sdf loading, for example.  I just want a subroutine to remove the meshlab step from this program.

sherm1, 2020-04-10:
@jwnimmer-tri thanks, got it. In that case I would propose we add this as a static method of the SpatialInertia class, similar to what we already have in the UnitInertia class:
[image]

So something like
```c++
static SpatialInertia<T> SpatialInertia::UniformDensitySolidFromSurfaceMesh(
    const std::filesystem::path& mesh_file_name, const T& mass);
```

jwnimmer-tri, 2020-04-10:
I'm not sure `SpatialInertia`, a fundamental value-type abstraction, is where I want my file opening and obj parsing to occur.  Seems like a layering violation.

sherm1, 2020-04-10:
I think it depends whether you see this as primarily 
- a mass properties operation (like we have for cubes, cylinders, etc.), or 
- a operation on meshes.

I don't see a layering issue or a logical issue as to why a SpatialInertia class wouldn't be a good place to calculate spatial inertias. Maybe as an optimization we would only want to open the mesh file and process it once? In that case we could provide an opportunistic geometry API that makes mass properties available from SceneGraph after it has processed the mesh. Even so I think SpatialInertia should be able to perform this computation, perhaps with a note saying that SceneGraph users can get it cheaper.

jwnimmer-tri, 2020-04-10:
I would buy this API for sure:
```
static SpatialInertia<T> SpatialInertia::UniformDensitySolidFromSurfaceMesh(
    const drake::SurfaceMesh& mesh, const T& mass);
```

But why should every single class in Drake that uses surface mesh data have APIs to read it from a file?  We should have an API to read from a file (possibly from many different formats!) into a value type, and an API to turn that value (+ mass) into an inertia.  Coupling the error-handling required for file parsing and reporting errors back to the user really pollutes the nice SpatialInertia data type.

sherm1, 2020-04-11:
Oh, yes that is much better. I like it.

DamrongGuoy, 2020-09-30:
I added label `component: geometry general`.  Please feel free to change it. I'm not sure whether it's `component: geometry proximity` or `component: multibody plant` too?

SeanCurtis-TRI, 2023-03-06:
This has arguably been resolved by #18345.

closed 2023-03-06 by SeanCurtis-TRI