RobotLocomotion/drake #13037: Can we push drake to Dockerhub

cjds, 2020-04-10:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)? yes
> * Have you reviewed existing GitHub issues and StackOverflow posts? yes
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?yes 

Can we make a Github Action that pushes a Dockerized version of drake to the central docker.io registry?

Ubuntu and ROS already do it, and Drake does have a Dockerfile inbuilt. 

https://hub.docker.com/_/ros

https://hub.docker.com/_/ubuntu

I'm willing to do the work, if this is something you are interested in

jwnimmer-tri, 2020-04-10:
Perhaps https://hub.docker.com/r/robotlocomotion/drake/tags has what you're looking for?  It appears to [not be mentioned on our website](https://drake.mit.edu/search.html?q=dockerhub&check_keywords=yes&area=default) though, so that's a bug.

jwnimmer-tri, 2020-04-10:
Oh, it's mentioned as "Docker Hub" not "dockerhub".  https://drake.mit.edu/docker.html#docker-hub

cjds, 2020-04-10:
That is indeed what I was looking for thanks

closed 2020-04-10 by cjds