RobotLocomotion/drake #13040: Drake limits a child body can only have one parent body

huihuaTRI, 2020-04-10:
##### Problem:
Given the robot defined  as the attached figure
[image]
, the user can only weld the `robot_root` to the ground. If the user chooses any other bodies, the MultibodyPlant (when Finalize) will complain:
**what():  This mobilizer is creating a closed loop since the outboard body already has an inboard mobilizer connected to it. If a physical loop is really needed, consider using a constraint instead. Aborted (core dumped)**

##### Diagnosis:
This error is because Drake requires a child body should only have one parent body, i.e., the definition of the robot has to be strictly a tree structure.
See some initial discussion here: https://tri-internal.slack.com/archives/C2BMF3K28/p1586557048162800

##### Feature request:
The MBP should support arbitrary topologies.

sherm1, 2020-04-11:
Assigning to @amcastro-tri initially for consideration (cc @joemasterjohn). This is _related_ to the bigger project of supporting general topologies but is actually much easier since the result is still a tree. It really only requires adding reverse mobilizers (in this case just reverse revolutes) so that the parent->child direction of the joints can be preserved even though inboard->outboard is reversed in the tree.

OTOH the general multibody graph code we have (at least in a branch of mine) for analyzing general loop topologies understands this kind of funky-tree topology also so it might be easier just to pull it in even if we still only allow trees.

EricCousineau-TRI, 2020-08-12:
From perusal (per Sherm's mention in #13859), I think it'd be important to mention that there's a workaround - just reverse the parent/child yourself in your spec?

Doesn't seem like that big of a deal IMO, but I can understand the error message being obtuse and being hard to debug, and being slightly annoying to specify it that way -- but it doesn't really seem that brittle?

Also, unclear to me: Does this relate #9566 (closed loop topologies)? And what's most important here?

huihuaTRI, 2020-08-12:
Reverse the parent/child may not necessarily be a solution for all cases. For example, given a multi-branch (Dr. Octopus) robot, you may want to weld different links to the world in different situations. The strict tree structure constraint will prevent the user from doing that.

It's different than a loop joint. This one constrains that you can have a maximum one parent link.

EricCousineau-TRI, 2020-08-12:
Ah, gotcha! Sorry, hard to disentangle the loop topology issue from reordering inboard/outboard, but I see - you wanna specify multiple "parents" in your spec, but have MBP resolve the inboard/outboard bits.

And I see - if mobilizer flipping is done to maintain the "World must be inboard" constraint, then at that point either specifying the "world" as either parent or child of any link should work.

sherm1, 2020-08-12:
The way this is handled in internal coordinate multibody codes is to use a "reverse mobilizer" internally to implement the user's joint. (This is one of those cases where the distinction between joint and mobilizer is significant, though still hidden from users.) That still leaves a tree structured model (no loops) but makes the signs of the joint coordinates match the user's expectation. For example, a reverse revolute mobilizer simply defines its generalized coordinates q and v with the opposite meaning from the usual revolute mobilizer. The MBP Finalize() code then chooses the appropriate mobilizer to implement the joint in MBTree.

mitiguy, 2020-09-02:
The inboard/outboard terminology and related issues (#12789 and #13040) arose in PR# 13961.
it would be helpful to agree/converge on frame names for documentation and notation.  
For example, we currently use Parent body P with frame F and Child body B with frame M.
Perhaps this should be Parent body P with frame Jp and Child body C with frame Jc.

amcastro-tri, 2020-09-03:
I've been consistently using P/F with B/M at the "mobilizer" level (internal tree model of our world) and P/Jp and B/Jc at the "joint" level (physical representation of our world). 
If somewhere in Drake you find an inconsistency, it should be fixed. But I believe we did like this in the past, given that mobilizers will eventually be reversible in order to enable the modeling of arbitrary loop kinematics.

sherm1, 2020-11-06:
Related: #9939

sherm1, 2022-06-21:
Closing this in favor of duplicate request #17429.

PR #22122 (2024-11-18): [multibody] Allow reversed weld joints
When building the tree-structured model for a system of rigid bodies and joints, it is possible that the inboard/outboard sense of the underlying mobilizer will have to be reversed from the parent/child ordering of the joint it models. MultibodyPlant currently forbids that; this PR relaxes that restriction for Weld joints, which are the most common case we see where reversal is needed.

Fixes #13040

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22122)
closed 2022-06-21 by sherm1