RobotLocomotion/drake #13041: Using drake as a third party Bazel project

cjds, 2020-04-11:
I'm trying to use Drake in a third party project that uses Bazel but there seem to be some issues that prevent me using it

This is my setup 

WORKSPACE
```
workspace(name="test_drake")
load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive")
http_archive(
    name = "drake",
    urls = ["https://github.com/RobotLocomotion/drake/releases/download/v0.16.1/drake-20200312-bionic.tar.gz"],
    sha256 = "b8bd38808d52340ea32f0c26d925b49dd2a79e85b80a4bc519553d825c144c67",
)
```

test_drake/BUILD (this is a copy of the rod example in the Drake project)

```

load(
    "@drake//tools/skylark:drake_cc.bzl",
    "drake_cc_binary",
    "drake_cc_googletest",
    "drake_cc_library",
)
load(
    "@drake//tools/vector_gen:vector_gen.bzl",
    "drake_cc_vector_gen_library",
)

package(default_visibility = ["//visibility:public"])

drake_cc_vector_gen_library(
    name = "rod2d_state_vector",
    srcs = ["rod2d_state_vector_named_vector.yaml"],
)

drake_cc_binary(
    name = "rod2d_sim",
    srcs = ["rod2d_sim.cc"],
    add_test_rule = 1,
    test_rule_args = [" --sim_duration=0.01"],
    deps = [
        ":rod2d",
        "@drake//attic/systems/rendering:drake_visualizer_client",
        "@drake//common:add_text_logging_gflags",
        "@drake//common:essential",
        "@drake//lcm",
        "@drake//lcmtypes:viewer",
        "@drake//systems/analysis",
        "@drake//systems/framework",
        "@drake//systems/lcm:lcm_pubsub_system",
        "@drake//systems/rendering:pose_aggregator",
        "@drake//systems/rendering:pose_bundle_to_draw_message",
        "@gflags",
    ],
)

drake_cc_library(
    name = "rod2d",
    srcs = ["rod2d.cc"],
    hdrs = [
        "rod2d.h",
    ],
    deps = [
        ":rod2d_state_vector",
        "@drake//common:essential",
        "@drake//multibody/constraint",
        "@drake//solvers:mathematical_program",
        "@drake//systems/framework:leaf_system",
        "@drake//systems/rendering:pose_vector",
    ],
)
```

This is the error that I get
```
ERROR: error loading package test_drake': Every .bzl file must have a corresponding package, but '@drake//tools/skylark:drake_cc.bzl' does not have one. Please create a BUILD file in the same or any parent directory. Note that this BUILD file does not need to do anything except exist.
INFO: Elapsed time: 0.048s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (1 packages loaded)
```

sherm1, 2020-04-11:
Assigning to @jamiesnape for a first look -- please reassign if appropriate.

jwnimmer-tri, 2020-04-11:
FYI https://github.com/RobotLocomotion/drake-external-examples/tree/master/drake_bazel_external serves as a template for how to do this.

What version of Bazel are you using?

cjds, 2020-04-11:
Version 2.1.1. Looking at the example I managed to get over this hump by changing the WORKSPACE to

```
load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive")
http_archive(
   name = "drake",
   urls = ["https://github.com/RobotLocomotion/drake/archive/v0.16.1.tar.gz"],
   sha256 = "f63940c274f90ffa4943d13865abc623455bec124c2dc15a9c8c9c4c67a209e1",
   strip_prefix="drake-0.16.1"
)
load("@drake//tools/workspace:default.bzl", "add_default_workspace")
add_default_workspace()
```

But I still get issues while trying to use `drake_cc_library` and `drake_cc_vector_gen_library` in my build. 

Specifically I get, 
```
ERROR: Analysis of target '//test_drake:rod2d_state_vector' failed; build aborted: no such package 'tools': BUILD file not found in any of the following directories. Add a BUILD file to a directory to mark it
as a package.

```

I noticed that in the referenced package you stay away from the custom drake_cc_libs and vector_gen_libs. Are you not supposed  to use those rules and use just refer to drake as a shared_library instead?

jwnimmer-tri, 2020-04-11:
Ah, I see now.

Declaring deps like` "@drake//common:essential"` is no problem.  You may use `:drake_shared_library` if you wish, but using the smaller targets is fine.

The problem is using `drake_cc_library` -- that's only intended for use within drake.  Downstream should use just `cc_library` or `cc_binary`.

https://github.com/RobotLocomotion/drake-external-examples/blob/master/drake_bazel_external/apps/BUILD.bazel#L36-L44

I should update the docs (and visibility) to make this more clear.

cjds, 2020-04-11:
I gotcha..Closing this based on above. Thanks

closed 2020-04-11 by cjds