RobotLocomotion/drake #13047: docker installation instructions may lead windows users astray.

RussTedrake, 2020-04-12:
Reported [on stack-overflow](https://stackoverflow.com/questions/61168362/drake-problem-first-installation-in-windows).

Our instructions on https://drake.mit.edu/docker.html point people to some out-of-date resources from the MIT classes.  I think we should move any tips for windows users into the drake docs, or somehow produce an issue/forum where we can let the community update our collective knowledge about if / how that works.  (it could even be a github issue, in which case the work would just be to write "windows is unsupported, but please contribute any of your updates/findings as a comment on the issue...").

jwnimmer-tri, 2020-05-14:
Relates #7654 (mostly a duplicate).  The bottom line is that we need to zap the confusing, stale, unsupported, and untested text, one way or another.  We should probably even do that ASAP, even if we don't have replacement text ready yet.

jwnimmer-tri, 2020-05-28:
Calling this a duplicate of #7654, we'll remove the instructions.

closed 2020-05-28 by jwnimmer-tri