RobotLocomotion/drake #13058: C++ API and pybind: Reconsider usage of unique_ptr

EricCousineau-TRI, 2020-04-14:
Broader issue which covers current issues with `unique_ptr` issues like:
* #12811 - unavoidable delete since C++ resets the memory without letting `pyinbd` ever access the holder
  * TODO(eric): Find reference to "snowflake pointer" proposal I made
* General confusion about lifetime semantics with Python bindings (e.g. the nuances I've attempted to expand upon in #11984)
* Bugs related to my implementation of `unique_ptr` in general:
  * #13056
  * TODO(eric): List more here.

Some more salient conversation about this starts here:
https://github.com/RobotLocomotion/drake/issues/12811#issuecomment-600931700

My concern is that if we go with @jwnimmer-tri's suggestion of making input argument `T*` overloads in C++ (to pair with `unique_ptr`) and expose them to Python, we are (a) making a *very* poor-man's copy of `shared_ptr` and (b) only take a few bullets out of the foot-gun (e.g. deletion in C++ could still completely invalidate a `T*` reference).

However, it would solve maybe most of the issues for now, until we permit more dynamic stuff (like possibly deleting things in MultibodyPlant (e.g. if #12703 ever implies deleting elements).

The perhaps longer-lasting solution would be to switch over to `shared_ptr`. On the pros, it can make Python bindings dramatically simpler (and thus easier to automate in #7889), perhaps enable us to remove some of our pybind fork's features. On the cons, there are painful reference cycles that are much harder to debug than with `unique_ptr`, perhaps performance (though maybe a more minor point?), and the fact that `shared_ptr` is greedy (cannot convert to `shared_ptr` due to its [patchy?] reentrancy).

FYI @jwnimmer-tri @RussTedrake @sherm1

sherm1, 2020-04-14:
Are you proposing changing Drake's C++ API to use shared_ptr? I would be strongly opposed to that since it would turn our current crisp ownership clarity into a fuzzy mess.

EricCousineau-TRI, 2020-04-15:
Yup, that is one proposition, and the fuzzy mess is what I alluded to.

I think it matters how smooth we want the overall experience to be, including scripted languages.

EricCousineau-TRI, 2020-04-15:
Sort-of: That is one proposition, and the fuzzy mess is what I alluded to.

The other proposition is what Jeremy suggested: to make special overloads with `T*` args instead of `unique_ptr<T>` on the C++ API, specifically for handling ownership issues with Python. That will alleviate some concerns, but only part-way (I can expand more if you like).

I guess the "third proposition" is do nothing and keep things as they are. I guess I like this more than the second proposition, because this one requires zero add'l effort (but has more likelihood for user errors).

I think it matters how smooth we want the overall experience to be, including scripted languages. I think now is the time we seriously consider how Python factors into the experience, rather than it being an afterthought - I am, of course, quite biased ;)

jwnimmer-tri, 2020-04-17:
For API design, I think its important to use concrete examples to evaluate potential alternatives.  Here, we'd identify places that we currently use unique_ptr on purpose, and see how they would look under any new regime.

I think #8904 is an obviously good change that we should just do everywhere (to simplify both C++ and Python life).

But for other, real cases of ownership transfer and ownership trees, we should collate several categories of use from the above list of issues and citations, expand it with any other ones found via grep, and use those specific APIs in specific example programs as the metric to discuss the alternatives.

In general, I would be happy if our solution ended up with only upstream pybind11's level of support for unique_ptr, without drake-specific ownership customizations.  But I think we need to walk through some examples to see how plausible that is.

EricCousineau-TRI, 2020-10-20:
Keep on forgetting where this is: #7715 is the snowflake `unique_ptr_tracked` thing

EricCousineau-TRI, 2020-11-18:
Came up again:
https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1605703451063400

jwnimmer-tri, 2022-02-25:
> In general, I would be happy if our solution ended up with only upstream pybind11's level of support for unique_ptr, without drake-specific ownership customizations. But I think we need to walk through some examples to see how plausible that is.

I did a quick skim today.  On first look, it does seem possible to unfork ourselves, and return to upstream pybind11.

We have many functions that return a `unique_ptr`.  This is already supported upstream in the obvious way (they become shared).

The places where in our public API have `unique_ptr` as an input are where the challenge comes in.

There seem to be a few broad categories where `unique_ptr` is used as input:

(1) The Simulator's handle to its contained System and Context.
(2) The subsystems with a Diagram (and thus, how DiagramBuilder works).
(3) The sub-elements (e.g., Parameters) within a Context.
(4) The MbT elements being added (AddBody, etc.)
(5) Wrapper classes in `systems/{controllers,estimators}`, e.g, `linear_model_predictive_controller`.

For (1) (2) (5), I think it's probably fine to just bite the bullet and switch to shared_ptr (with appropriate deprecation).

For (3), we should see how much of the `unique_ptr` API surface actually needs to appear in pydrake.  Most of it is probably framework-internal calls only.

For (4), I haven't investigated yet.

EricCousineau-TRI, 2022-02-25:
Re: upstream merging, there is also support for `unique_ptr<>` as input:
https://github.com/pybind/pybind11/blob/smart_holder/README_smart_holder.rst
It is not strictly upstream per se, going to be better than our current implementation.

Been meaning to integrate it, but has been low on my queue, e.g. https://github.com/RobotLocomotion/pybind11/pull/56

jwnimmer-tri, 2025-01-09:
As part of#21968 we're basically working on this now, so I've moved this over to Rico's attention.

rpoyner-tri, 2025-03-18:
As of #22738, we are back on vanilla pybind (with a few patches not related to permitting unique_ptr arguments). I will claim that unique_ptr has been fully reconsidered.