RobotLocomotion/drake #13064: API to specify point contact parameters.

amcastro-tri, 2020-04-15:
Thus far our point contact model has been used as a penalty method to enforce regularized non-penetration and friction constraints.
For stiffness, our constants are global for the entire model, based on "penetration allowance".
Unfortunately nowadays "penetration allowance" has become a proxy for tinkering with compliance given the lack of other options.

We'd like "penetration allowance" to *guide* an initial set of parameters so that users do not need to worry about it for rigid geometries. However, we'd like to allow users to change stiffness on a per-geometry basis so that they can, for instance, model compliant components in their model.

Before implementing this we'd need to settle on:
1. What parameters we'd like to expose? for instance, if we have a normal force law of the form `fₙ = k ⋅ (1 + d ẋ)₊ ⋅ (xⁿ)₊` (with the usual symbols x for penetration, (x)₊ the positive part of x, d dissipation, etc), would we expose `k`, `d` and `n`?
2. How would we name those parameters in the user-facing API?
3. What are the SDF/URDF tags we'd like to provide?
4. Implementation. I'd imagine this would go through `ProximityProperties`?

Looking for input to resolve points 1-4 before action. 
cc'ing @sherm1, @SeanCurtis-TRI, @joemasterjohn, @naveenoid, @sammy-tri, @ggould-tri

SeanCurtis-TRI, 2020-04-15:
:+1: on going through `ProximityProperties`.  From a parsing perspective, we'd dump them in :`<drake:proximity_properties>`.

Once we figure out which parameters we're exposing and what we're calling them (particularly with respect to how they play with the parameters we're currently using for hydroelastic contact), we'll figure exactly how they're stored in proximity properties.

ggould-tri, 2020-04-15:
At low confidence:
1. 
 * Do you mean (xⁿ)₊ or (x₊)ⁿ ?  For even n the distinction seems important.
 * Otherwise this seems fine -- gives an easy division into the conservative and nonconservative components.
 * I do worry a bit that k ("normal force when stationary at 1m penetration distance") is going to be too large a quantity to reason about.  Is this conventional?
2.
 * k has units of N/(m^n); it's not going to have a nice conventional name with that dimension, but conceptually it is "normal force at 1m penetration".  So something with no specific meaning but indicative like "penetration force constant"?
 * d has units of s/m; dimensionally I think it is "strain per unit velocity", since it's a strain augmentation to the normal force.  But that's silly and unintuitive; "dissipation" is a reasonable name for it.  Or "penetration drag" since drag has no conventional dimension or units anyway.
 * n is just an exponent.  Call it "penetration exponent" or something.
3. The SDF/URDF tags should match the API parameter names as much as possible.
4. Yes.

amcastro-tri, 2020-04-15:
1.
Great catch on (x₊)ⁿ @ggould-tri, updated comment.
Is this conventional? yes, think of conventional linear spring, for which k has units of k/m.

2. 
- k has units of N/(m^n)...
Good point. I believe we'd like a default value of `n=1` for the reason you mention. However, some users wanting to experiment with Hertz law might want to play with that exponent. For sphere contact, the parameters are k = 4/3 E* √R and n = 3/2, with E* the effective Hertz Young modulus and R the (effective) sphere radius. So units do make sense for this case also.
- d has units of s/m;
Correct. It's the Hunt & Crossley dissipation model. 
- Call it "penetration exponent"
maybe. What about `point_contact_law_exponent`? with default value of 1.0.

sherm1, 2020-04-15:
I would advocate not overthinking these parameters! Point contact is not a great model of physical contact. Users will generally be tweaking these parameters in terms of stiffer/less stiff and more and less dissipation. They just need some easy-to-turn knobs. How about:
- k: stiffness in units of force/distance
- d: dissipation in units of 1/velocity

I don't think providing an exponent is helpful. Perhaps we could allow specification of default values for k and d, with the default-default being something like stiff rubber or nylon-ish.

@amcastro-tri I think the attempts to have the code guess these parameters makes this too complicated to explain and for users to reason about. I'd rather see a simple default and then documentation on how to estimate reasonable values. Any users have an opinion on that?

amcastro-tri, 2020-04-15:
@sherm1, we'd then need to deprecate already available functionality. This proposal is just an add-on to existing functionality. I'd avoid deprecating stuff.
The defaults parameters are very easy to document, "set_penetration_allowance()" already documents this behavior. 

On the subject of the exponent, is a nice-to-have feature for more experimental users like myself wondering the implication on continuity. But willing to give it up if no one likes it.

sherm1, 2020-04-15:
> On the subject of the exponent, is a nice-to-have feature for more experimental users like myself

I agree but it's awkward for users because it changes the units for k. Maybe we can still do it internally without exposing it to users?

sherm1, 2020-04-15:
> The defaults parameters are very easy to document, "set_penetration_allowance()" already documents this behavior.

But now we will have to explain how this interacts with specified k,d right? Is that straightforward?

amcastro-tri, 2020-04-16:
yes. That idea would be to leave the penetration allowance be the default as it is now. Then the new APIs would allow users to change those defaults.

joemasterjohn, 2020-07-16:
Resolved by #13630 and #13686.

closed 2020-07-16 by joemasterjohn