RobotLocomotion/drake #13069: Integrators should reset cached matrices when user changes settings

antequ, 2020-04-16:
Currently, if a user creates an Implicit Integrator (e.g. Implicit Euler Integrator), runs it, and then decides to change the Jacobian computation scheme, the implicit integrator does not discard the cached Jacobian. If the user continues to integrate after this, it will reuse the cached Jacobian until it decides that it needs to recompute it. If the user wanted the integrator to reset the memory, he needs to call Initialize() again to reset the cached Jacobian and iteration matrix. This applies to all the memory managed by the integrators that is cached and reused. (Note: this data does not use the systems' caching system.)

This was pointed out by the code change and discussion in PR #12967 , in implicit_integrator_test.h .

In general, integrators should discard or mark its cached matrices as dirty when the user changes settings that affect the cached data.

antequ, 2020-04-16:
It turns out this is a bug that I introduced only to VelocityImplicitEulerIntegrator. ImplicitIntegrator itself actually discards its own Jacobian when the integration scheme changes. I believe this also forces it to recompute iteration matrices during the first step. Happy to close this if there is nothing else that could be an issue.

antequ, 2020-04-16:
We basically want this line:
https://github.com/RobotLocomotion/drake/blob/580795905a580f78e7b856ee5d4b80822504dc61/systems/analysis/implicit_integrator.h#L116

to also reset the Jacobian in VelocityImplicitEulerIntegrator.

Two possible solutions are
1) add a do_reset_cached_jacobian() method to ImplicitIntegrator, called from set_jacobian_computation_scheme() that subclasses can override (this would get us the same behavior as other implicit integrators), or
2) cache the previously used computation scheme in VelocityImplicitEulerIntegrator, so that when it changes, we recalculate the Jacobian. This will be checked in MaybeFreshenVelocityMatrices(), which is where deleting the cached Jacobian would trigger a Jacobian recomputation anyways.

The difference in behavior between the two routes is, if the user changes the Jacobian computation scheme and then changes it back, BEFORE calling another step, do we want it to discard the cached Jacobian? That is what it would do for other implicit integrators. If so, solution (1) is the best route.

sherm1, 2020-04-16:
My preference would be to wipe out any saved state whenever a potentially-invalidating change occurs, in some rote manner like a call to WipeOutAnySavedState(). (So essentially (1) above.) I wouldn't want to see the Jacobian logic get any more complicated! We definitely shouldn't optimize for the change-it-and-then-change-it-back scenario.

closed 2020-05-30 by antequ