RobotLocomotion/drake #13074: mbp: Support programmatic composition and decomposition among "sub-plants"?

EricCousineau-TRI, 2020-04-16:
## Composition

It would be nice to be able to assemble different plants together, i.e. throw them both into the same world, and then throw in some welds.
This would most likely require #12703 (the topology variant).

## Decomposition

This would cover the solution I mentioned here: https://github.com/RobotLocomotion/drake/issues/12980#issuecomment-614940839
Basically, take an existing mega-plant (e.g. simulation world), and identify sub-graphs, and generate a physically valid `MultibodyPlant` from that.

## Use Cases

* Computing for subsets (as mentioned in #12980)
* Being able to make IIWA controllers without excessive "parameter lumping" (kinematic and inertial)
  * See example in Anzu (can maybe post cut-out of code)
  * See example in `ManipulationStation`: https://github.com/RobotLocomotion/drake/blob/d1e11190794d8b31be5ad5918c8d0f30db7b2874/examples/manipulation_station/manipulation_station.cc#L420-L450

sherm1, 2020-04-16:
@EricCousineau-TRI a few thoughts:
- I'm imagining that you would need also to transfer part of the larger plant's state to the sub-plant's state. Is that part of the decomposition task?
- These seem like two unrelated tasks. Reading in multiple models and welding them together before doing anything with them seems very easy. Extracting sub-pieces of a plant, determining that they form a coherent connected set, and possibly transferring state seems MUCH harder. If I've understood that correctly please consider filing separate issues so we can prioritize and scope separately.

ggould-tri, 2020-04-17:
This is a really nice idea -- it would go a long way toward cutting the knot on #12849 .  However as @sherm1 notes it would also incur all of the complexity that motivates the 12849 case -- the correct shuffling and interpretation of state and the implementation of state readback into the parent -- and it would lay a serious landmine under any logic that uses fortran style "offset into flat vector" logic to address mbp states (ie, nearly all of it :sob: :rage: ).

EricCousineau-TRI, 2020-04-29:
Sorry for the lag! I will try to break this apart as requested by EOD today.

EricCousineau-TRI, 2020-04-29:
Will update this issue to handle the harder task (decomposition). Will make a new issue for the easier task (composition).
Will edit this post once I'm done.

EricCousineau-TRI, 2020-04-30:
MATLAB version: https://www.mathworks.com/help/robotics/ref/rigidbodytree.addsubtree.html

sherm1, 2020-05-11:
Lowering priority for now since this was a lesser issue for users per the Mutating Multibody Models meeting.

EricCousineau-TRI, 2020-05-11:
Per convo, also should be achievable by current public API, without having to a method on `MultibodyPlant`:
[Drake Dev Meeting Agenda, 2020/05/11](https://docs.google.com/document/d/15iNBw_2O3TGbThHr6wlJNKctHSY5bEvt1rLhl5obZao/edit#)

EricCousineau-TRI, 2020-05-16:
I am working on a prototype of this in Python, as part of #13133 

Similar to what Robert ran into, I'm seeing parsing taking up at least 50% of my team for clutter gen. I want to see if I can cut that down by some sort of subgraph decomposition and assembly and stuff.

EricCousineau-TRI, 2020-05-19:
I think I have a prototype in Anzu that covers the workflows I want. See here for a copy-pasta version:
[`MultibodyPlantSubgraph`](https://github.com/EricCousineau-TRI/repro/tree/master/drake_stuff/multibody_hacks) ([permalink](https://github.com/EricCousineau-TRI/repro/tree/d22602d781b569d7996850878222a56197cfc762/drake_stuff/multibody_hacks))

This can handle composition via sub-plants.
Currently, the Anzu PR is in place, and will let it bake for a few weeks there per Jeremy's suggestion.

EricCousineau-TRI, 2021-01-06:
See user pain point that could be partially addressed by this: https://stackoverflow.com/questions/65583969/visualizing-an-autodiff-multibodyplant-in-pydrake#comment115962920_65584221

jwnimmer-tri, 2026-09-15:
It is difficult to tell what feature is actually being requested here, but in any case we don't plan to work on it.

closed 2026-09-15 by jwnimmer-tri