RobotLocomotion/drake #13082: proposal: Change implementation of min for AutoDiffScalar when args are equivalent

avalenzu, 2020-04-17:
Eigen's implementations of `min` and `max` for `AutoDiffScalar` have internal inconsistencies in which argument they return if `x == y`. Less troubling perhaps, but still a potential pitfall, is that they don't always follow the same convention as `std::min` and `std::max`.

***Please note before proceeding, that everything is fine if `x == y` is `false`. The issues only arise for the case where `x == y` returns `true`. For `AutoDiffScalar` variables this means that `x.value() == y.value()`.***

What follows is @jwnimmer-tri's description (lightly edited from a Slack discussion) of the current status and his proposal (with which I concur) for how to resolve the situation:

Let ADS/0 denote an `AutoDiffScalar` with `derivatives().size() == 0` at runtime, and ADS/N denote an `AutoDiffScalar` with `derivatives().size() != 0` at runtime.

There are four functions under consideration: 
- Eigen `min`
- Eigen `max`
- `std::min`
- `std::max`

Each function takes two arguments, `a` and `b`, in that order.

Each argument has three options: double, ADS/0, ADS/N.  So that's a 3x3 grid of cases, for each of the four functions:

|  a \ b | double | ADS/0 | ADS/N
|---------------|----------|----------|----------|
| **double** |  * | * | #
| **ADS/0** | * | * | ?
| **ADS/N** | #  |  ? | ?!

If we have both arguments as any combination of `double` or ADS/0, then all four functions agree on what to return -- the `double` value converted to ADS/0 (or left as `double` in the case where both `a` and `b` are `double`. This covers the 2x2 grid of [`double`, ADS/0]^2 marked with "*" above. The question is what to do with ADS/N.

For ADS/N op ADS/N (marked with "?!" above), we have Eigen `max`, `std::min`, `std::max` all returning `a`, but Eigen `min` returning `b`.  I see no reason why that one should differ from `std`, so I think **we should replace it**.  I don't think we need min + max to be mirrors of each other or somesuch?  It seems weird.

I think the `double` op ADS/N (marked with "#" above) that Eigen already defines (always return the ADS/N argument) seem sensible for heterogenous types.  When the types are homogeneous I think std's semantics wins, but for mixing types I think preserving the derivs when equal no matter which side they came from is fine.

Therefore, and finally, I think we should treat ADS/0 op ADS/N (marked with "?" above) as-if the ADS/0 were double, and always return the ADS/N argument. We will have to be careful that the min and max overloads for ADS are always used, and not std's template.  I think that's fairly practical.

All of that together, the only changes are to the following cases:
- Eigen min(ADS/N, ADS/0)
- Eigen max(ADS/0, ADS/N)
- Eigen min(ADS/N, ADS/N)

avalenzu, 2020-04-17:
cc @jwnimmer-tri. Could you take a look and make any edits you deem necesssary before I publicize this more widely?

jwnimmer-tri, 2020-04-17:
Looks fine to me.  I will just post the current tables (when a == b) for reference.

I also realized that, relatedly, we should double-check that Expression's min & max operators match the std's semantics.

---

std::min and std::max are just
|  a \ b | double | ADS/0 | ADS/N
|---------------|----------|----------|----------|
| **double** | a | a | a
| **ADS/0** | a | a | a
| **ADS/N** | a | a | a

---

Eigen min is currently:
|  a \ b | double | ADS/0 | ADS/N
|---------------|----------|----------|----------|
| **double** | - | b | b
| **ADS/0** | a | b | b
| **ADS/N** | a | b | b

---

Eigen max is currently:
|  a \ b | double | ADS/0 | ADS/N
|---------------|----------|----------|----------|
| **double** | - | b | b
| **ADS/0** | a | a | a
| **ADS/N** | a | a | a

avalenzu, 2020-04-17:
`Expression`'s implementations do match `std`:
https://github.com/RobotLocomotion/drake/blob/master/common/symbolic_expression.cc#L733-L737
https://github.com/RobotLocomotion/drake/blob/master/common/symbolic_expression.cc#L745-L749

avalenzu, 2020-04-17:
@RussTedrake, @sherm1, @soonho-tri, @hongkai-dai (and anyone else who's interested): Please take a look at this proposal and weigh in.

sherm1, 2020-04-17:
I am in favor of
(a) making these consistent and preserving derivatives as proposed, and
(b) not writing code that depends on that behavior!

Is (b) possible? This seems like a very brittle dependency and likely to cause bugs, esp. down the road when a new programmer does a little "harmless" tidying up.

One case that's not covered here is `min/max(ADS/N, ADS/M)` where M ≠ N and neither is 0. Can that happen, and if so what should be the behavior? (abort seems reasonable!)

jwnimmer-tri, 2020-04-17:
When M != N, many of the current overloads for other binary math operators would hit an Eigen size mismatch abort iff in Debug mode, and that seems good here too.  I'd also be okay with ignoring it, treating it as an unchecked precondition.  If your ADS derivatives are ever different-and-nonzero then something is seriously wrong, no matter what operation is underway.

I think the important part of (b) is not _implicitly_ depending on the "first argument wins" behavior.  If the clearest way to write the code is ADL `min` and `max` plus a Chesterton's-fence comment reminder the reader that the first argument's derivatives wins, then so be it.

sherm1, 2020-04-17:
TIL Chesterson's Fence! I love it; can't wait to find a use for it myself.

SeanCurtis-TRI, 2020-04-19:
Likewise TIL -- and I've *read* Chesterton.

avalenzu, 2020-04-20:
If no one has objections by EOD today (2020-04-20), I think we should go ahead with this. Last call, @RussTedrake, @soonho-tri, @hongkai-dai.

soonho-tri, 2020-04-20:
SGTM. Could you also report this to the upstream?

RussTedrake, 2020-04-20:
Just took a few minutes with it.  I agree with your assessment.  SGTM.

avalenzu, 2020-04-21:
After working on this a bit, I'm concerned about the level of complexity we're introducing into what ought to be a simple operation. Compare an implementation of the proposed changes to `min`:

https://github.com/avalenzu/drake/blob/2e2d2f5c67b4a257f464b77c79c2770805b5225f/common/autodiffxd.h#L421-L448

to what would be needed to make it behave like `std::min` (and Eigen's `max`):
```.c++
inline const AutoDiffScalar<VectorXd> min(const AutoDiffScalar<VectorXd>& a,
                                          const AutoDiffScalar<VectorXd>& b) {
   return (a <= b) ? a : b;
}

avalenzu, 2020-04-21:
Thanks, that's much more manageable (and it passes the same tests I put in place for my longer version).

PR #13112 (2020-04-22): common: Add implementations of {min,max}(AutoDiffXd, AutoDiffXd)
Resolves #13082.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13112)
commit 79a2c8dc (2020-04-22): common: Add implementations of {min,max}(AutoDiffXd, AutoDiffXd) (#13112)

Resolves #13082.
closed 2020-04-22 by avalenzu