RobotLocomotion/drake #13087: VTK python wrapper incompatible version

zixiliu, 2020-04-18:
I'm on Mac OS Catalina 10.15.4. 
When I build and test by _bazel build //..._ and _bazel test //..._, the build is successful and the tests all pass. 
However, when I run the drake visualizer by _./bazel-bin/tools/drake_visualizer_, I get the following error message: 

> dyld: Library not loaded: /usr/local/opt/vtk@8.2/lib/libvtkWrappingPython37Core-8.2.1.dylib
  Referenced from: /Users/zixiliu/my_git_repos/drake/bazel-bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/bin/drake-visualizer
  Reason: image not found
Abort trap: 6

During installing, I ran _./setup/mac/install_prereqs.sh_ and verified both Python3.8 and vtk8.2 are installed.

Running _bazel clean_ and reinstalling the prereqs did not resolve the issue. 

Git log hash: 68169e6e13616f6057729c949d79e0f53fa5528f

See below is the output to _brew linkage vtk@8.2_. It looks like while the program is looking for a python3.7 wrapper, only python3.8 wrapper exist for vtk. 

> System libraries:
  /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
  /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa
  /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
  /System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics
  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
  /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
  /usr/lib/libSystem.B.dylib
  /usr/lib/libc++.1.dylib
  /usr/lib/libexpat.1.dylib
  /usr/lib/libobjc.A.dylib
  /usr/lib/libxml2.2.dylib
  /usr/lib/libz.1.dylib
Homebrew libraries:
  /usr/local/opt/freetype/lib/libfreetype.6.dylib (freetype)
  /usr/local/opt/glew/lib/libGLEW.2.1.dylib (glew)
  /usr/local/opt/hdf5/lib/libhdf5.200.dylib (hdf5)
  /usr/local/opt/hdf5/lib/libhdf5_hl.200.dylib (hdf5)
  /usr/local/opt/jpeg/lib/libjpeg.9.dylib (jpeg)
  /usr/local/opt/jsoncpp/lib/libjsoncpp.21.dylib (jsoncpp)
  /usr/local/opt/libogg/lib/libogg.0.dylib (libogg)
  /usr/local/opt/libpng/lib/libpng16.16.dylib (libpng)
  /usr/local/opt/libtiff/lib/libtiff.5.dylib (libtiff)
  /usr/local/opt/lz4/lib/liblz4.1.dylib (lz4)
  /usr/local/opt/netcdf/lib/libnetcdf.18.dylib (netcdf)
  /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (qt)
  /usr/local/opt/qt/lib/QtGui.framework/Versions/5/QtGui (qt)
  /usr/local/opt/qt/lib/QtSql.framework/Versions/5/QtSql (qt)
  /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets (qt)
  /usr/local/opt/ospray@1.8/lib/libospray.0.dylib (robotlocomotion/director/ospray@1.8)
  /usr/local/opt/ospray@1.8/lib/libospray_common.0.dylib (robotlocomotion/director/ospray@1.8)
  /usr/local/opt/vtk@8.2/lib/libvtkChartsCore-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkChartsCorePython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonColor-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonColorPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonComputationalGeometry-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonComputationalGeometryPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonCore-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonCorePython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonDataModel-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonDataModelPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonExecutionModel-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonExecutionModelPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMath-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMathPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMisc-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMiscPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonSystem-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonSystemPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonTransforms-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonTransformsPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDICOMParser-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistry-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
[... 175 lines cut ...]
  /usr/local/opt/vtk@8.2/lib/libvtkViewsContext2DPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkViewsCore-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkViewsCorePython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkViewsInfovis-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkViewsInfovisPython38D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkWrappingPython38Core-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkdoubleconversion-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkexodusII-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkgl2ps-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtklibharu-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtklzma-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkmetaio-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkproj-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkpugixml-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtksqlite-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtksys-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkverdict-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/szip/lib/libsz.2.dylib (szip)
  /usr/local/opt/tbb/lib/libtbb.dylib (tbb)
  /usr/local/opt/tbb/lib/libtbbmalloc.dylib (tbb)
  /usr/local/opt/theora/lib/libtheoradec.1.dylib (theora)
  /usr/local/opt/theora/lib/libtheoraenc.1.dylib (theora)
Indirect dependencies with linkage:
  szip
  tbb

soonho-tri, 2020-04-18:
> During installing, I ran ./setup/mac/install_prereqs.sh and verified both Python3.8 and vtk8.2 are installed.

Please checkout the latest `master` and rerun the `install_prereqs.sh` script. It will use `/usr/local/opt/vtk@8.2/lib/libvtkWrappingPython38Core-8.2.1.dylib`.

zixiliu, 2020-04-19:
Confirming working now. Thanks a lot!

closed 2020-04-19 by zixiliu

jwnimmer-tri, 2020-04-20:
FYI There wasn't a good issue filed yet about the overall change; I've added #13087 now for that purpose.