RobotLocomotion/drake #13099: Documentation for multi-threaded use of Drake

jwnimmer-tri, 2020-04-20:
It's not clear from the Drake docs how to best use Drake in a multi-threaded situation.

For example, at TRI we often use a single MultibodyPlant (i.e., single Diagram) with multiple separate diagram Contexts, and use one Context per thread to compute from different states concurrently, but sharing the single plant.

When multiple threads want to share the same Context, the code should first prime it and then [FreezeCache](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context_base.html#a72d7aa9c0de7f2d0b18fc5dde3e199b0) and [UnfreezeCache](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context_base.html#abb3e9df7a99f4d1b7d6aaefdadac28cd).

Classes that are an exception to these rules have special warnings documentation, e.g. [SignalLogger](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_signal_logger.html).

We should try to consolidate this advice into a Doxygen section to pull all of the information together.

See also  #10383, #10228, #10496.

\CC @sherm1

jwnimmer-tri, 2021-07-12:
The documentation in RobotLocomotion/styleguide/46 begins this journey, but doesn't mention everything, and isn't in Doxygen.

\CC @rpoyner-tri

rpoyner-tri, 2026-11-12:
Reassigning to @sherm1, the `system framework` lead per the component table in GitHub Issue Management. I've done no work on this, and multithreaded use of Systems and Contexts is squarely in that component.

@sherm1, please keep it, relabel it, or close it as you see fit. I'm happy to review a docs PR.

closed 2026-11-13 by rpoyner-tri