RobotLocomotion/drake #13102: Ubuntu 20.04 (Focal Fossa) support

jwnimmer-tri, 2020-04-20:
This is an anchor issue for discussing goals and timelines for 20.04 support.

\CC @jamiesnape

jwnimmer-tri, 2020-04-20:
My starting bid for Drake would be 20.04 release date (scheduled for late April) + 60 days for having nighty binary releases ready. The mid-July Drake release would be the first candidate for supporting 20.04.

I don't know of any high-priory use cases yet, so we could slip some weeks or a couple months if other priorities take over.

jamiesnape, 2020-04-21:
Probably removing the semi-artificial error messages and getting an `install_prereqs` and `Dockerfile` up are the first steps. I tried the beta out earlier and there weren't major issues with basis provisioning, though vtk and director would still need rebuilding due to the python version difference.

jamiesnape, 2020-04-21:
`libnetcdf13`, `libyaml-cpp0.5v5`, `libhdf5-100` are the package name changes. `kcov-35` is not going to available since we build and host that ourselves, but they now have `kcov` proper since `eoan`, so that is good.

jwnimmer-tri, 2020-04-27:
My thoughts on next steps:

(1) Define a few (initially non-Production) CI configurations for 20.04.  I think I'd be satisfied with only:
  - linux-focal-gcc-bazel-continuous-release
  - linux-focal-gcc-bazel-experimental-release
  - linux-focal-unprovisioned-gcc-bazel-nightly-release
  - linux-focal-unprovisioned-gcc-bazel-experimental-release

Once those are working and we're closer to turning on official support, we can chase down the rest of the full matrix.

Should I open a new issue for this?

(2) Implement #10729 so the visualizer releases are serviceable by others, and then deploy the 20.04 vtk and director builds.

(3) Iterate on `build ...` and `test ...` passing.

Depending on what the landscape for (3) is looking like, we can figure out how to get a good subset of progress under CI or not.

jamiesnape, 2020-04-27:
Probably just the unprovisioned flavors (we can have still have unprovisioned continuous) are fine while we iterate and have only two or three jobs, but please create an issue. One complication is the Java version that is available with Focal. I will look into it and comment on the new issue if it is a problem.

jamiesnape, 2020-05-04:
Is my understanding correct that we will need `python-is-python3` as it stands for Bazel to be happy (and if we do, that we may want `python-dev-is-python3` for consistency to avoid the odd nasty surprise down the road)?

jwnimmer-tri, 2020-05-04:
My WIP testing has the below added to `setup/ubuntu/source_distribution/install_prereqs.sh`

We should not force people to have `/usr/bin/python` be set to 3, but we should force them to set it to something.

```bash
if [[ "${codename}" == 'focal' ]]; then
  # We need a working /usr/bin/python (of any version).  On Bionic it's there
  # by default, but on Focal we have to ask for it.
  if [[ ! -e /usr/bin/python ]]; then
    apt-get install --no-install-recommends python-is-python3
  else
    echo "/usr/bin/python is already installed"
  fi
fi
```

I was going to wait for Focal in CI before PRing that.

I haven't checked if `python-dev-is-python3` is required.  I'd hope out toolchains could be sure to choose the `python3-dev` without the metapackage.

jamiesnape, 2020-05-04:
> We should not force people to have /usr/bin/python be set to 3, but we should force them to set it to something.

Agreed.

`python-dev-is-python3` is pretty much just setting `python-config` to match `python`. It is not required by us, but we do use `"{}-config".format(/path/to/python3)` somewhere, I think.

jamiesnape, 2020-05-04:
Need to add doxygen to the install_prereqs (and maybe update the version on Xenial to match).

jamiesnape, 2020-05-13:
I am seeing several broken `.pc` files in Focal so far.

jamiesnape, 2020-05-13:
For one of `.pc` file issues, the frustrating thing is that I fixed the file in `yaml-cpp` (https://github.com/jbeder/yaml-cpp/pull/556) and then Debian or Ubuntu applied a patch that broke it again (https://salsa.debian.org/debian/yaml-cpp/-/blob/master/debian/patches/install-cmake-dev-files.patch).

jamiesnape, 2020-05-20:
Possibly hitting https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80635.

jamiesnape, 2020-05-20:
GCC 9 really does not like `IsMemcpyMovable`.

jamiesnape, 2020-05-26:
First yellow build: https://drake-jenkins.csail.mit.edu/job/linux-focal-unprovisioned-gcc-bazel-experimental-release/55/ (https://github.com/RobotLocomotion/drake/pull/13385)

jamiesnape, 2020-05-26:
Looks like the clang part of the python documentation extraction/transformation is failing (though not causing a build error, slightly worryingly).

jamiesnape, 2020-06-02:
GCC debug builds are broken with at least one `-Werror=sign-compare`.

jamiesnape, 2020-06-02:
Apparently GCC 9 dislikes the comparison operators in `TypeSafeIndex`.

jwnimmer-tri, 2020-06-08:
@jamiesnape Modulo the open issues and PRs already shown upthread in this issue, could you clarify for me what (if any) planned work remains here?

It seems to me like the only remaining item is to move some of the focal gcc builds into the https://drake-jenkins.csail.mit.edu/view/Production/ label to be buildcop'd.

jamiesnape, 2020-06-08:
Adding at least one build to the list of pre-merge checks would probably be the only other thing once we have run the continuous in production for a week or so.

jwnimmer-tri, 2020-06-08:
SGTM.

jwnimmer-tri, 2020-06-11:
@jamiesnape I think we're ready to move focal into Continuous Production and add one pre-merge now, if you agree?

Since we plan to have clang-9 parity already on bionic and focal, I'll suggest to add one of the faster gcc variants, maybe `linux-focal-gcc-bazel-experimental-release`?  We can keep the clang pre-merge on bionic for easier local repro for current developers.

jamiesnape, 2020-06-11:
Done.

closed 2020-06-11 by jamiesnape