RobotLocomotion/drake #13103: Should document (un)supported features for sim/multibody description formats

EricCousineau-TRI, 2020-04-20:
Motivated by:
* @jwnimmer-tri's [Slack post](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1587402931061500)
* [Drake Developers Meeting Agenda, April 13, 2020](https://docs.google.com/document/d/15iNBw_2O3TGbThHr6wlJNKctHSY5bEvt1rLhl5obZao/edit#heading=h.y0xdhn25sh16)

Should have:
* Table of (un)supported features for SDFormat
  * I would like to include a "roadmap" of sorts for things that are under development.
* Table of (un)supported features for URDF

These tables should have indication for collision types as well.

\cc @SeanCurtis-TRI

SeanCurtis-TRI, 2020-04-20:
This should also include any custom `drake:`-namespaced tags that we parse.

EricCousineau-TRI, 2020-04-20:
For collision, should also consider pairwise collision constraints as indicated in #10907.

jwnimmer-tri, 2022-05-02:
I believe that @rpoyner-tri has already accomplished this in documentation:

https://drake.mit.edu/doxygen_cxx/group__multibody__parsing.html

@EricCousineau-TRI can you take a look and let us know what else is missing?

It doesn't have a roadmap, but _shrug_ that doesn't seem terribly important.  We can use the issues database for tracking any requests that are acute enough to merit action.

EricCousineau-TRI, 2022-05-03:
Looks good to me! In the future, perhaps we can say which portion of SDFormat we do(n't) parse (e.g. we don't parse `//actor`, `//link/enabled_wind`), but if no one's asking atm, we can defer until acute need arises.

Thanks!

closed 2022-05-03 by EricCousineau-TRI