RobotLocomotion/drake #13116: Add contact results visualization to tutorials/rendering_multibody_plant.ipynb

RussTedrake, 2020-04-22:
We don't show anybody, yet, how to visualize forces.  It would be easy/nice to add to the tutorial.

So far, when someone asked, I've responded with:

In meshcat, we have:
https://github.com/RobotLocomotion/drake/search?q=MeshcatContactVisualizer&unscoped_q=MeshcatContactVisualizer

For drake-visualizer, we have:
https://github.com/RobotLocomotion/drake/search?q=ConnectContactResultsToDrakeVisualizer&unscoped_q=ConnectContactResultsToDrakeVisualizer

jwnimmer-tri, 2022-05-06:
The basics of collision display probably do deserve a mention in the rendering tutorial.

We've also talked about adding a dedicated tutorial for collisions / proximity, which should incorporate a comprehensive look at all kinds of collision visualization.

jwnimmer-tri, 2022-08-03:
Note that #17690 is somewhat related, but has a different focus.

jwnimmer-tri, 2023-09-23:
Now that we pervasively suggest `AddDefaultVisualization`, this mostly works correctly automatically.  What remains here is perhaps to add more explanatory text, or a deeper dive on understanding what we're showing when we see collisions in the default visualization.

closed 2024-03-16 by jwnimmer-tri