RobotLocomotion/drake #13117: The binary generated by Cmake build and Bazel build are different.

mjm522, 2020-04-22:
When I perform the Bazel build following the instructions in the site, I get all the plugins activated by default. This is the window of the Bazel built visualizer.

[image]

However, the Cmake build version and the Binary provided shows only partial number of plugins. This is the window of the Cmake built visualizer. The provided Binary also looks the same.

[image]

I am looking to ways of activating the contact plugins. I tired passing the script to the `drake-visualizer` in the binary folder with the `--script` flag as follows:

```
drake-visualizer --script ~/drake/tools/workspace/drake_visualizer/plugin/use_builtin_scripts.py
```
But this resulted in some path error.

Could someone please suggest how to activate all plugins in the binaries or how to the cmake build with all the plugins activated?

jwnimmer-tri, 2020-04-23:
Apologies for the trouble.  See #10486 for the (still unfixed!) existing bug report on this topic.  The problem is the same -- only bazel builds have the plugins enabled.

I'll see if I can post a work-around into that issue.

closed 2020-04-23 by jwnimmer-tri

mjm522, 2020-04-23:
Using the info given by @jwnimmer-tri in  [#10486](https://github.com/RobotLocomotion/drake/issues/10486) , I copied the plugin folder to the bin and it works. The repo is [here](https://github.com/mjm522/drake-binary-with-plugins). This is a quick fix for anyone using the binary to get all the plugins activated.

You need to run the `drake-binary-with-plugins/bin/drakeVisualizerPluginEnabled`