RobotLocomotion/drake #13122: Issues setting contact model parameters in URDF or programmatically

gpldecha, 2020-04-23:
Hi,

I am a new user of Drake. I am having issues being able to set the dynamic properties of rigid objects in pydrake. 

At the moment the coulomb friction parameters do not seem to be parsed from a URDF.
I posted on [stackoverflow](https://stackoverflow.com/questions/61367504/pydrake-urdf-setting-dynamic-parameters) the example URDF. If I am missing something or doing something wrong here, any help would be appreciated (I showed how to set them programmatically in the answer).

On a second point, I have been in formed that the default contact model is "Penalty method point contact". I would like to know if it is possible to set the "penetration" and "stiction" parameters in the URDF (although it has been indicated that this might not be supported yet). 

The alternative would be to set them programmatically. However after printing the available functions of the MultibodyPlant object the following two functions:

* set_penetration_allowance
* set_stiction_tolerance 

were not exposed. Any help on this would also be appreciated. 

On a side note, I think an example in python which shows how to manipulate (set/get/urdf/sdf) the parameters of the different contact models would be nice.

SeanCurtis-TRI, 2020-04-23:
Hi @gpldecha, thanks for the issue.

1. Coulomb friction: thanks for the example. Once you change `<drake:drake>` to `<drake:proximitiy_properties>` the coulomb friction *should* parse. (You don't need and shouldn't use `<drake_compliance>`.) As pointed out in the stack overflow, there is no discoverable documentation on what gets parsed and what doesn't in URDF/SDF.

2. Penalty method: See issue #13064 for a discussion on setting contact parameters on a per-object basis. But you are correct, the penalty model is configured *on* the MultibodyPlant in code and not via URDF parsing. Looking at the code it seems that `set_penetration_allowance` [is bound](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/multibody/plant_py.cc#L706-L708). However, `set_stiction_tolerance` appears not to be.

I'm going to leave this issue open in order to get the binding to get updated. But we already have issues about the discoverability of parsing functionality (#13124 and #13103).

gpldecha, 2020-04-24:
Hi @SeanCurtis-TRI, thanks for the reply.
Yes, I see that set_penetration_allowance is accessible. I will watch out for   
set_stiction_tolerance in a new update :+1:

SeanCurtis-TRI, 2020-05-13:
@amcastro-tri I'm passing this to you and tagging it with MBP because the only thing remaining here is the binding of an MBP method.

amcastro-tri, 2020-05-18:
Resolved in #13294.

closed 2020-05-18 by amcastro-tri