RobotLocomotion/drake #13124: Parsing documentation for friction has broken links

amcastro-tri, 2020-04-23:
Per [SO discussion](https://stackoverflow.com/questions/61367504/pydrake-urdf-setting-dynamic-parameters/), the docs on how to set friction parameters are not discoverable.

I found out the only one point of entry to them is in the MBP [docs here](https://github.com/RobotLocomotion/drake/blob/d24bc88e50e3a506cfcff98bed2f88350f34243e/multibody/plant/multibody_plant.h#L1242). Not only the link doesn't quite belong there (or I should rather say, it should be moved up to the `mbp_contact_modeling` section since it applies to both contact models)

Still, the but is that these docs live in `internal::` and do not show up in Doxygen. The links above are broken.

SeanCurtis-TRI, 2020-04-23:
Closing as being redundant. #13103 covers this.

closed 2020-04-23 by SeanCurtis-TRI

amcastro-tri, 2020-04-23:
I don't think it is redundant. The bug I describe is not about writing new documents, but about somehow moving what already is in there outside the `internal::` namespace. Hopefully an easier task?