RobotLocomotion/drake #13126: Can't programmatically alter transparency of model loaded from URDF/SDF in visualization

SeanCurtis-TRI, 2020-04-23:
A user requested the ability to visualize a robot with transparency. The robot is specified in a URDF/SDF file, which in turn references objs and mtl files. The challenge, is that the user wants to apply transparency *on top* of those specifications. In other words, they shouldn't have to change the specification on disk to achieve a run-time effect.

The relationship between `IllustrationProperties` and `drake_visualizer` is such that if the ("phong", "diffuse") value has a non-unit alpha channel, that alpha value will *still* affect the visualization of the robot whose visual appearance would *otherwise* be defined by the mtl file. This provides the means.

The missing piece is that `SceneGraph::AssignRole()` isn't implemented for `IllustrationProperty` and `RoleAssign.kReplace` so the diffuse value *cannot* be changed after parsing. We need to implement this in order to enable this behavior.

SeanCurtis-TRI, 2020-04-23:
There's a reason it hasn't been implemented yet. It's discussed in this TODO: 

https://github.com/RobotLocomotion/drake/blob/master/geometry/geometry_state.cc#L784-L789

tl;dr Changing the illustration properties does not guarantee that illustration engines will reload their geometry. So, if the load message has already been sent to drake visualizer and *then* illustration properties are sent, no visible difference will appear.  

The short-term solution is to simply document this and put the burden on the caller to make sure they make changes to illustration properties *before* they connect visualizers.

mposa, 2020-04-24:
As the user above, I approve this feature request!

This also seems relevant for https://github.com/RobotLocomotion/drake/issues/9728

PR #13598 (2020-07-08): [geometry] IllustrationProperties can be updated after parsing
This PR has two curated commits:

1. The first extends `GeometryProperties` functionality to make it easier to modify properties. A mutable instance of `GeometryProperties` can have properties removed, updated, or overwritten.
2. The second enables `AssignRole(..., IllustrationProperties, AssignRole::kReplace)`. Previously, that would throw. Due to issues in the greater ecosystem, this added functionality comes with limitations and warnings.
  - there are warnings in the documentation.
  - Warnings will likewise get printed to the  [... cut]
closed 2020-07-08 by SeanCurtis-TRI