RobotLocomotion/drake #13131: mutable "other_system_context" pattern is now common; breaks thread safety.

RussTedrake, 2020-04-24:
Our standing recommendation for thread-safety is "with a few notable exceptions (like `SignalLogger`), if you use one `Context` per thread, you should be safe".  But we have a pattern that has become common, and is difficult to avoid, where we have a mutable class member that stores a context for another system that it needs to call methods on.

Probably the most minimal and best documented example is [LuenbergerObserver](https://github.com/RobotLocomotion/drake/blob/master/systems/estimators/luenberger_observer.h#L67).  But it happens in `InverseDynamics`, `MultibodyPositionToGeometryPose`, `RiccatiSystem`, `Propeller`, ...

Just to put it out there, here are a few broad-strokes approaches (some of which I know will get shot down hard):
1. for kinematics in particular, add a KinematicsQuery port to MBP.  but that doesn't help e.g. the `InverseDynamics` use case, and is probably not a general enough solution.
2. Keep using the pattern we have, admit that we aren't thread safe.
3. Keep using the pattern we have, but throw in a few mutex calls inside the methods.  I've always thought this would be pretty simple, but remember a comment from @jwnimmer-tri before about how (almost) every use of mutex, etc, is a defect.
4. Come up with a way to register another system's context as a part of my own context, without it massively polluting everything.  (For instance, the `Propeller` system should have zero state; I don't want it to suddenly have to declare the entire state of MBP, or some impenetrable AbstractState).  This is not state, it's like a scratchpad of intermediate variables.  Perhaps we need a mechanism for this in `Context`?

avalenzu, 2020-04-24:
> 4. Come up with a way to register another system's context as a part of my own context, without it massively polluting everything. (For instance, the Propeller system should have zero state; I don't [...]

Could it be a cache entry?

SeanCurtis-TRI, 2020-04-24:
Random idea:

What if it were defined as a `static thread_local` variable of the calc function that uses it?

SeanCurtis-TRI, 2020-04-24:
Random idea:

What if it were defined as a `thread_local` variable of the calc function that uses it? Or if it needs to be shared across functions, declared as `thread_local` in its accessor and every other function accessess it that way?

jwnimmer-tri, 2020-04-24:
(Thread-local doesn't work -- you need per-System-instance-per-thread.  There's more than one instance of a given System.)

The need for a per-context allocated scratchpad is widespread.  I believe we've already discussed it in several other issues; I'll try to find citations later on.

SeanCurtis-TRI, 2020-04-24:
Yeah....you're right. Not a general solution.

sherm1, 2020-04-24:
To make sure I understand the problem: is the thread-unsafety caused by there being a mutable member in what should be a const System? If so it does seem like a sound solution would be to have that mutable memory located in the per-thread Context instead.

As @avalenzu points out above, a cache entry is already a chunk of mutable memory in the Context, capable of storing any type-erased object. If it's declared with `{nothing_ticket()}` as its sole prerequisite it will just sit there forever holding its current contents. If that's too ugly we could slap some sugar on top and add a first-class notion of scratch memory in the Context.

jwnimmer-tri, 2020-04-24:
@sherm1 I'd thought in the past you'd strongly argued that cache entries should not be used as scratchpads?

sherm1, 2020-04-24:
I have always felt strongly that cache entries should have predictable persistence from call to call (I depend on that in Simbody, for example, to manage expensive state variables like delay tables). But I do kind of remember some controversy over that here that I think maybe we never resolved.

But assuming they are guaranteed persistent (which is the de facto case now anyway) they _could_ be used as scratch memory. That's not part of the original intent for cache entries, but I can't think of anything fundamentally wrong with using a no-dependencies cache entry as scratch. Overblown API for that simple purpose but could be sugared easily: `DeclareCachedScratchMemory()` could make a cache entry with no Calc() method and no dependencies.

RussTedrake, 2020-04-25:
Thanks all.  I will take a spin at moving `LuenbergerObserver` to using a cache entry; though perhaps not immediately.

edrumwri, 2020-04-25:
> But assuming they are guaranteed persistent (which is the de facto case now anyway) they _could_ be used as scratch memory. That's not part of the original intent for cache entries, but I can't [...]

FYI, we have been successfully using cache entries for workspace variables. I didn't actually think there was a big need for this outside of our group, since it's usually fine- though perhaps not fully performant- to declare variables on the heap.  Without studying @RussTedrake's application, it's unclear to me why declaring on the heap wouldn't work. Perhaps it's because of an implicit desire among Drake developers to avoid heap allocations for performance reasons? In any case, I appreciate the result: I have thus far found it to be pretty straightforward to avoid heap allocations when using the systems framework and various Drake systems. 

I'll note that the ID controller is not real-time safe at the moment, so the ID controller would satisfy neither real-time nor one-Context-per-thread applications.

RussTedrake, 2020-05-14:
I've got another interesting case in this pattern.  I don't believe that any of these systems support scalar conversion, yet?  I would like to write
```
  // Scalar-converting copy constructor.  See @ref system_scalar_conversion.
  template <typename U>
  explicit MySystem(const MySystem<U>&) : MySystem<T>(owned_system.ConvertOrThrow<U>(), ...) {}
```
I could certainly specialize the template and call `ToAutoDiffXd` and `ToSymbolic` myself, but that seems brittle (and we are talking here about a common pattern, not a one-off).  It looks like I could do roughly what `ToAutoDiffXd` is doing, but that feels like I'm calling internal methods.  @jwnimmer-tri -- what do you think?

jwnimmer-tri, 2020-05-14:
I think [get_system_scalar_converter](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#a686f497f27885d9f632df0c43d29a0d6) is the basis of what you'd want.  Something like `owned_system.get_system_scalar_converter().Convert<U, T>(owned_system)`.

The problem is that we've crept beyond the converter having all the answers, so you'd lose any external constraints:
https://github.com/RobotLocomotion/drake/blob/3e70123bd8cfbeebba57aa513531383d5e52d1cb/systems/framework/system.cc#L757-L767

It should be possible to either move the constraint transmog into the convert functors directly and then use the converter object directly yourself, or else offer a `System<U>::ToScalarTOrThrow<T>` method that calls one of the other two methods via `if constexpr`, now that we only support a closed set of scalar types.

RussTedrake, 2020-05-14:
Right.  I saw that code in my reading.  My current systems don't have external constraints, but I don't want to establish a brittle pattern.  I think offering a `ToScalarOrThrow<T>`might be best?  (note that I removed the T from the name, since it defeats my perception system and I can't help but read "to scalar to", plus I think the T is implied).

jwnimmer-tri, 2020-05-14:
Adding a templated `To...` converter is fine.  Ideally, we'd offer all three flavors (static, maybe, throws) and use analogous spellings to the current AutoDiff and Symbolic methods, for consistency.  (Assuming the static one's more-templated still makes sense; I haven't penciled it out.)

The converter is already public so it would be nice if its Convert method was correct, but we can save that for another day.

amcastro-tri, 2020-05-21:
> MySystem(const MySystem<U>&) : MySystem<T>(owned_system.ConvertOrThrow<U>(), ...) {}

I fell into this patter myself today. Concretely, I have a FooSystem that ideally would have access to all of MBP's APIs. Say I go around it by making my FooSystem to own a (cloned) plant, what about the context for that plant when I need to compute something? the system then would need to own its own plant context? what about caching? in this case there is an "original" MBP in the same diagram and ideally I'd like to use the same caching.
Sorry if not related, but I believe it is? any ideas?

jwnimmer-tri, 2020-05-21:
See #13307.

sherm1, 2020-05-21:
> When both systems co-exist in a single Diagram, perhaps we can provide a mechanism to ask for const access to other_system's Context.

Although this is somewhat dangerous, I think it addresses many practical cases where access to a peer subsystem's API is needed. (The "clean" way is to add a large assortment of very-narrow customizable output ports to the peer; that is a major yak shave for making a simple API call.) Peer access could be made fully transmogrifiable in the usual way by storing a SubsystemIndex rather than a pointer to the peer subsystem (which is today's very common practice). It would be very easy to add a few Diagram APIs to enable this, e.g. `my_subsystem.GetPeerSubsystem(peer_index)` and `GetPeerSubcontext(const Context& my_subcontext, peer_index)`.

sherm1, 2020-06-01:
@RussTedrake is using the cache to hold a mutable "other" context good enough that we can close this issue with that as a best practice? Or do we need to provide something else?

RussTedrake, 2020-06-02:
I think the pattern used in the updated Luenberger observer is the right pattern, but it still has a major limitation.  We cannot (in practice) perform scalar conversion on a system implemented using this pattern, until we resolve #13313.

I am ok if we close this given this breadcrumb, but #13313 is important.

closed 2020-06-02 by RussTedrake