RobotLocomotion/drake #13139: Need a `ValueMultiplexer` or at very least `ExternalAppliedSpatialForceMultiplexer`

RussTedrake, 2020-04-25:
Carry-over from #12401, where we observed that there is only one "applied_spatial_force" input port on MultibodyPlant, which takes a `std::vector<ExternalAppliedSpatialForce<T>>`.  But some use cases will require multiple external forces, coming from different systems, to be applied to the same MBP.

The `ExternalAppliedSpatialForceMultiplexer` is an obvious solution.  But one could also do a nice, fully-type-erased implementation of a more generic `ValueMultiplexer` that multiplexes multiple `std::vector<Value>` into a single one at the output.  As usual, @jwnimmer-tri had some excellent thoughts on #12401.  Opening this issue so that we don't loose track of it.

akshitj1, 2020-06-05:
Since anyone will need to add another plant eg. atleast some aerodynamic lifting sufrace with propller. Here is quick hack for non generic `ExternalSpatialForceMultiplexer` till generic `ValueMultiplexer` arrives: [gist](https://gist.github.com/akshitj1/01091bbc1599f0b3b0832efd74243fcc). Not raising a PR as I am not familiar enough with the framework to write tests.

jwnimmer-tri, 2022-10-24:
See https://github.com/RobotLocomotion/drake/issues/16923#issuecomment-1096837625.

I makes sense to add a class `ExternallyAppliedSpatialForceMultiplexer`, at least until the framework provides for something better.  However, Trying to make the `ExternallyAppliedSpatialForce` into a template argument of a generic `ConcatenateStdVector` template system will have too many sharp edges (as detailed in that discussion).

PR #18171 (2022-10-26): Add ExternallyAppliedSpatialForceMultiplexer
Resolves #13139
Relates #16923

py plant:
- Remove unneeded qualifier for ExternallyAppliedSpatialForce
- test_propeller: Use body name rather than magic number

adder_test:
- Add check for discrete number of states
- Use std::make_unique instead of .reset()

\cc @amcastro-tri @jwnimmer-tri 

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18171)
closed 2022-10-26 by EricCousineau-TRI