RobotLocomotion/drake #13146: Users can't infer the output port type from the MBP documentation

RussTedrake, 2020-04-26:
I had a question from someone who didn't understand why they could attach a `SignalLogger` to the `MultibodyPlant.get_generalized_contact_forces_output_port()`, things work, but connecting to `MultibodyPlant.get_reaction_forces_output_port()` or `MultibodyPlant.get_contact_results_output_port()`, it fails with

```
SystemExit: Failure at systems/primitives/signal_log.cc:14 in SignalLog(): condition 'input_size > 0' failed.
```

I see two defects here:  1) there is nothing in the MBP doxygen that would tell people what type to expect if they pull on the port.  2) `SignalLogger` should give a more helpful error message.

sherm1, 2020-05-01:
@amcastro-tri something for your miscellaneous-little-MBP-PRs list?

amcastro-tri, 2020-05-01:
I can definitely update the MBP docs. Maybe a different issue for `SignalLogger`?

jwnimmer-tri, 2024-03-20:
See also https://github.com/RobotLocomotion/drake/issues/9393#issuecomment-419060972 about the unclear semantics (beyond just the data type).