RobotLocomotion/drake #13162: py geometry: Issues when attempting to make Rviz visualizer

EricCousineau-TRI, 2020-04-28:
These are issues I ran into when trying to make a demo Rviz Python application, that I would like to be fixed:
https://github.com/RobotLocomotion/drake/issues/9500#issuecomment-620722987

- [ ] Missing C++ API (and bindings) for `X_FG`
- [ ] Missing bindings for `X_WG`, `X_WF`, some Shape attributes, and `SceneGraphInspector.get_all_frames()`
- [ ] Unable to use `GeometryProperties.GetProperty("phong", "diffuse", ...)` in Python, b/c it uses `Value<Vector4d>`. Need to think of how to expose this, or just register the concrete type(s) for related vectors

FYI @SeanCurtis-TRI @mntan3

SeanCurtis-TRI, 2020-04-28:
You can check the X_FG. It is not missing. I had brainfart when we were discussing it. X_FG is *not* dependent on inputs. It's a fixed parameter. Therefore, it is available from the `SceneGraphInspector::GetPoseInFrame()`.

EricCousineau-TRI, 2020-04-28:
Ah, good point. Was just trying to find the function `X_FG` - thanks!

EricCousineau-TRI, 2020-04-28:
@SeanCurtis-TRI Regarding the `GetProperty("phong", "diffuse", ...)` stuff, I would like to see if there's a way to simplify this type-erasure for vectors.

Can you foresee *only* using fixed-size vectors with `Value<>`? If so, I can figure out how to register those in such a way that isn't too ugly in Python.
Alternatively, we can go the route that `InputPort.Eval` goes, which is to explicitly separate numeric type erasure from `AbstractValue` type erasure (which is what Jeremy implemented):
https://github.com/RobotLocomotion/drake/blob/89767c81fe8c063d20194daeab1eb7c361069156/bindings/pydrake/systems/framework_py_semantics.cc#L29-L46

FTR, I would like to avoid needing `ExtractPhongProperties()` (and even `MakePhongProperties`) in Python. It's weird-feeling indirection.

@jwnimmer-tri Any chance you have opinions / suggestions on the matter?

SeanCurtis-TRI, 2020-04-28:
The `GeometryProperties` was devised such that *anything* that can go into a `Value` can go into `GeometryProperties`. The point is that `SceneGraph` can't forsee what could go in there. I can anticipate the kind of things that `SceneGraph` might want in there to consume. But any system can associate any data with their geometry such that they can pull it out during calculations.

Artificially shrinking the defining set pushes out the wall. But, conceivable, that wall will get hit.

As to your specific question: do you mean that the only types of value stored in geometry properties will be fixed-size vectors? Or the only vectors stored will be fixed size. Big no on the former and see previous paragraph on the latter. So, I lean towards the latter solution. More or less.

Currently, we have a number of scalar-valued properties in use. Are they strictly compatible with a Vector1? But I can imagine that we could do a number of things where we expand the overloads to include more metadata about the properties being stored. Something that would be invisible to the average C++ user but would facilitate python bindings.

EricCousineau-TRI, 2020-04-28:
> The GeometryProperties was devised such that anything that can go into a Value can go into GeometryProperties.  [...]

To clarify, my current issue is that this value handling does not have parity with the Systems framework (split numeric and abstract to avoid needing to care), meaning that we have to solve this problem again here. (Then again, this is kinda admitted by the fact that `Value<>` is permissible for Eigen types without constraint?) 

That being said, your later responses does indicate you're open to constraining it - woot!

>  So, I lean towards the latter solution. More or less. [...]

To be sure I understand, you meant that the vectors will have a constant size, like in the Systems framework; e.g. backed by `VectorX`, but never permitted to have a different size?

> Currently, we have a number of scalar-valued properties in use. Are they strictly compatible with a Vector1?

My vote on this is a huge *no*. If it's a vector, it's a vector, period.
Just type `[scalar]` in Python or whatever in C++. Scalar promotion to vectors are really weird in places like NumPy, and I'd love to not have that problem here (or at least, admit it on purpose).

On that point, I really want to ensure that we nip this problem in the bud with "best practices", at least inside Drake (but ideally, for downstream too via enforcement). e.g.
* If you want vectors or matrices, you have to use `VectorX<>` or `MatrixX<>`.
* If you *really* need fixed-size vectors, wrap it in a struct.
* Textures, as specified in Drake, should *only* be Drake image types, and *never* `Matrix<>`.

*Meta*: Lemme know if you want me to re-scope this issue to focus just on this problem, or make a new one. I can put remaining Rviz stuff elsewhere.

\cc @sherm1 as this may also imply some Systems stuff.

SeanCurtis-TRI, 2020-04-28:
Ah...I hadn't thought that "fixed-size" was a subset of the `kNumeric` treatment of ports, but, of course, you're right.

While there's a part of my soul that is loathe to let go of "if it goes in `Value` it goes in `GeometryProperties`", I'm willing to give up dynamically sized eigen types. What else would I have to give up to go that route? Do we still get non-vector-valued properties? Can someone just pull out a double? How much of a hit is the C++ functionality going to take?

> My vote on this is a huge no.

I hadn't realized it was a vote. I was just pointing out that we already have tons of code that uses the scalars. If we lose that, it'll be a heck of a flag day (and far clunkier call sites).

> On that point, I really want to ensure...

On what point? The scalar promotion you vetoed and I never proposed?

However, it does seem like you're somewhat answering my main question: what are the constraints on `GeometryProperties` that you're proposing?

(I do think the subject of "how do we bind geometry properties" is deserving of its own issue.)

EricCousineau-TRI, 2020-04-28:
Currently discuss in VC.

EricCousineau-TRI, 2020-04-28:
Per f2f wwith @SeanCurtis-TRI and @jwnimmer-tri, we're gucci on:
* Restricting to dynamic-only `VectorX` and `MatrixX`
* Introduce shims to still allow `VectorN` and `MatrixN`, but use dynamic-sized stuff in the actual `AbtractValue`
* Possibly deprecate call sites.
* *Maybe* introduce a color struct for RGBA (but also update downstream usages?)

sherm1, 2020-04-28:
> \cc @sherm1 as this may also imply some Systems stuff.

I'm not yet seeing how this affects System. I like the flexibility of GeometryProperties but don't see any problem with saying vectors have to be stored in VectorX's rather than Vector3's if that makes Python world nicer.

EricCousineau-TRI, 2020-04-28:
Sorry, meant that it may affect Systems as a consumer of `AbstractValue` if we develop guidelines / restrictions on public API considerations for vectors.
And sweet, sounds good!

EricCousineau-TRI, 2020-07-17:
Given that #10521 is it's own issue, will go ahead and close this given Mark's PR.

closed 2020-07-17 by EricCousineau-TRI