RobotLocomotion/drake #13165: [pydrake] Redundant event declaration for LeafSystem

SeanCurtis-TRI, 2020-04-28:
The binding of LeafSystem binds [several methods](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/systems/framework_py_systems.cc#L81-L86) for declaring events:
  
  - `DeclareInitializationEvent`
  - `DeclarePeriodicEvent`
  - `DeclarePerStepEvent`

Use in python tends to look like this:

```py
        self.DeclareInitializationEvent(
            event=PublishEvent(
                trigger_type=TriggerType.kInitialization,
                callback=self._initialize))
```

This particular function does *one* thing for you -- it confirms that you've set the `trigger_type` to `TriggerType.kInitialization`. But for that, this would essentially be a generic event declaration function that  consumes the same event. The `DeclarePerStepEvent` is exactly the same.

__Issue 1__: My initial impression is that we're introducing API into pydrake that doesn't carry its weight. The putative value is slight. It's real value is if one is declaring an event with an `Event` instance of unknown provenance. The function call suggests expectation and it validates that expectation. However, that workflow seems rather rare.

__Issue 2__: The presence of these three invites the question, what about the other trigger types? The presence of these implies the presence of "Forced", "Timed", and "Witness". But they are conspicuously absent.

Perhaps it would be better to just have `DeclareEvent()` and call it good.

Alternatively, we have more elaborate sugar available in C++ which, if bound, would make that same example look like:

```py
        self.DeclareInitializationPublishEvent(publish=self._initialize)
```

This version seems to contribute meaningfully to the cause (although there is rather an explosion of these as we combine event type and trigger type.)

__Arguments against changing__

The `DeclarePeriodicEvent` actually has additional, required parameters (offset and period) and does interesting things internally. This clearly brings more to the table.

The presence of the methods leaves the door open for the implementations to possibly do more under the hood if the hidden implementation details need to change. That expansion would be much harder if people weren't already using semantically-targeted APIs.

EricCousineau-TRI, 2020-12-02:
Returning here: FWIW I'd argue that Python shouldn't be the only culprit for this issue. Having fewer C++ methods is always nice too ;)

jwnimmer-tri, 2024-03-16:
Seems to be fixed in #16251 or thereabouts.

closed 2024-03-16 by jwnimmer-tri