RobotLocomotion/drake #13168: Remove and/or refactor unnecessary, unused, and optional prerequisites

jamiesnape, 2020-04-29:
@RussTedrake and I have been having a few issues with the size of Drake in terms of disk space (especially the Docker image) when using it on GitHub Actions (regular Azure Pipelines users would have issues too). One way to easily address this would be to audit and slim down the prerequisites.

For instance, with some configuration changes, we can remove some of the VTK system dependencies (such as ogg and theora, which will happen naturally when rewrite the build scripts for focal) and remove or (most likely) refactor into different scripts optional tools, such as virtualenv, fonts, compilers, cmake.

Relates https://github.com/RobotLocomotion/drake/issues/12783, since Debian has concepts of "build-depends", "depends", "recommends", "suggests" for prerequisites that are useful.

jwnimmer-tri, 2020-04-29:
I've thought about this from time to time, also.

In the cases you mention, is it the binary prereqs or source prereqs (or both) that are in scope and causing trouble?

For source prereqs, I've been considering splitting out those necessary for different use cases:
- for users: install from source via drake_{bazel,cmake}_external
- plus for contributors: bazel test //... from source in the default configuration (default compiler, release, etc.)
- plus for core team: all of the extra stuff for CI variants (valgrind, coverage, etc.)

jamiesnape, 2020-04-29:
Acutely, the binary prereqs, but for packaging we ideally need to isolate the build depends correctly (which falls under source prereqs), so I would say both is in scope.

Separating the python dependencies and the development vs. runtime dependencies of the binary packages is similar issue.

Some of the CI may follow from infrastructure we would need for testing packages, but yes, it is not trivial to test.

jwnimmer-tri, 2020-05-15:
In particular, if we can get `boto3` out of the critical compilation path it would have helped with #12747.

jamiesnape, 2020-07-16:
Putting aside `drake-visualizer`, at a minimum we are going to want to split dependencies as follows

* *runtime*
* *development* (depends on runtime)
* *python* (depends on runtime)
* *build*
* *maintainer*

In theory, *runtime*, *development*, and *python* are a subset of `binary_distribution` and *build* and *maintainer* are subsets of the union of `binary_distribution` and `source_distribution` (life is not quite that simple as it stands).

jamiesnape, 2020-07-16:
Getting *runtime* and *python* correct are the priorities so that `apt install python3-drake` functions as expected.

jamiesnape, 2020-07-20:
My current thinking is that it might be a good idea to keep track of dependencies (and any other metadata, such as licenses) of each subdirectory of `tools/workspace` and then collate them into lists.

jamiesnape, 2020-07-23:
My ideal would be something along the lines of this, say in JSON (which would be accompanied by documentation, schema, and validation):
```
{
    "name": "drake",   # not strictly necessary
    "components": [
       "build", 
       "development",
       "java",
       "python",
       "runtime",
    ],
    "type": "source",   # or binary
    "dependencies": {   # maybe could be split Mac/Ubuntu/Bionic/Focal
        "build": {
            "apt": [],
            "brew": [],
            "cask": [],
            "pip": []
        },
        "runtime": {
            "apt": [],
            "brew": [],
            "cask": [],
            "pip": []
        },
        "development": {
            "apt": [],
            "brew": [],
            "cask": [],
            "pip": []
        },
        "java": {
            "apt": [],
            "brew": [],
            "cask": [],
            "pip": []
        },
        "python": {
            "apt": [],
            "brew": [],
            "cask": [],
            "pip": []
        }
    },
    "copyright": {
        "files": [   # type source only
            "*"
        ],
        "licenses": [
            "BSD-3-Clause"
        ],
        "copyright": [   # required type source, optional type binary
            "2012 Massachusetts Institute of Technology",
            "2016 Toyota Research Institute"
        ]
    }
}
```

JSON is nice because does not need any extra libraries to parse with Python and schemas are easy to write. Not quite so nice for humans. Other options are fragments of Debian-style `control` and `copyright` files, something like `Gemfile` or `Brewfile` (but Python instead of Ruby), TOML like Pipfile, ...

jamiesnape, 2020-07-23:
Metadata files with just the copyright parts in tree would be nice too, but that is for a different issue.

jwnimmer-tri, 2020-07-27:
Per #12783, with a single large deb package, this refactoring is slightly less urgent.  We can pursue the per-workspace json manifests down the road, but for now it seems to me like splitting the txt files once more (for source vs developer dependencies) should suffice and not require new technology.

jamiesnape, 2020-07-31:
For CI, we are going to provision the CMake builds and the packaging builds with only build dependences, so corresponding to `--without-test-only` in #13753. Maintainer dependencies, i.e., `-with-maintainer-only` will get a partial test with the S3 mirroring job. We probably could run the new release scripts for informational purposes so that they are exercised by CI as well.

jwnimmer-tri, 2020-08-06:
Next step is to update the CI configs to toggle these new options so we have some test coverage.  To start that will be only the unprovisioned variants, which are easy.  Later can be the provisioned AMIs -- that requires a bit more refactor of Jenkins stuff.

jwnimmer-tri, 2020-09-03:
Update f2f -- no real need to have provisioned AMIs for the multiple flavors, so we'll just to unprovisioned.

closed 2020-09-17 by jamiesnape