RobotLocomotion/drake #13176: scene_graph, query_object: Could provide easy query for when geoemtry changes?

EricCousineau-TRI, 2020-04-29:
See discussion here: https://github.com/ericcousineau-tri/repro/pull/1#pullrequestreview-403034482

Relates #12703, #13162

Main intended application is for visualizers that don't (currently) support geometry changes (i.e. adding a cylinder or changing its color) to have an easy way to fail fast if change is detected in geometry.
Secondary application would be to allow applications that do support geometry changes to simply "reset" everything, or compute its own set of deltas.

One solution is to have a serial number of sorts, and let the consumer (system or code) remember the last-used serial number (e.g. `QueryObject::get_geometry_change_id`, which could just increment based on mutation - not hash).
If it changes, then that implies change.
The system itself would be responsible for doing things correctly (e.g. using cache), or doing some sense of undeclared state (#4330).

My hope is to maybe have something like this within the next few months (September?), if possible?

\cc @sherm1 @amcastro-tri If y'all think this may somehow relate to how "invalidation" would be performed for MBP mutations?

jwnimmer-tri, 2020-04-29:
The key question is what's covered by the token, and what is not.  The the only thing not covered supposed to be re-posing the geometries?  If proximity geometries change but the illustration geometries do not change, is that a new token?  It seems very application specific, and difficult to generalize.

If the only purpose is for failing fast, then performance does not matter, I think?  You could re-reify (brute force walk) the query at 1 Hz at low amortized cost, I think?

sherm1, 2020-04-30:
A serial number for any non-pose geometry change seems harmless and possibly useful. That could be extended to add finer-resolution serial numbers ("appearance change number") as needed. I'm more curious about how the changes get delivered to the visualizer though. Does SG push a "reinit" message? Does the visualizer have to ask for the new geometry after it finds out there's a change?

jwnimmer-tri, 2020-04-30:
The downstream consumer system would just read the token (ticket) at the same time it reads the poses.  Then it would know if it can re-transmit the new poses safely, or if something else has happened in the meantime.

sherm1, 2020-04-30:
> The downstream consumer system would just read the token (ticket) at the same time it reads the poses. Then it would know if it can re-transmit the new poses safely, or if something else has happened in the meantime.

I got that part but am wondering what happens next?

SeanCurtis-TRI, 2020-04-30:
If' it's a visualizer, it basically dispatches the new version of the "load world" message it did upon initialization (by trolling the SceneGraphInspector's data), followed up by the "pose world" message that is now compatible with the world.

It's not as nice as a mechanism said, "oh, this one small thing changed", but it allows the visualization to keep up with arbitrary changes.

SeanCurtis-TRI, 2020-04-30:
Or, in the fail fast sense, it can say, "Hey! You changed things. I can't do that. Stop it!"

sherm1, 2020-04-30:
The visualizer would be the final recipient of the "load world" message. So is it the intermediate message-generating subsystem that would have to watch for a serial number change and initiate a new "load world" message on its own?

SeanCurtis-TRI, 2020-04-30:
Exactly. The visualizer is running in a separate process (most likely). The "intermediate message-generating subsystem" is, in fact, that last downstream object in the Drake process that can do that. 

In my mind, distinguishing the two from a Drake process perspective doesn't have a lot of value. Drake knows nothing about the other process and *can* know nothing about the other process. It only knows whether it's going to spew messages, what types, and under what conditions. Even if they only disappear into the ether, the act of spewing messages is everything.

PR #14196 (2020-10-23): Add GeometryRevision.
Resolves #13176. 

GeometryState owns a GeometryRevision that enables a easy query when the underlying geometries change.
Downstream systems may query the revision numbers through SceneGraphInspector.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14196)
commit 81aebc7a (2020-10-23): Add GeometryRevision. (#14196)

* Add GeometryVersion.

Resolves #13176. GeometryState owns a GeometryVersion that enables a
easy query when the roles of the underlying geometries change.
Downstream systems may query the version through SceneGraphInspector.
closed 2020-10-23 by xuchenhan-tri